ESPresense/ESPresense #11: Flashing via web browser not working

ragenhe, 2021-09-07:
The process does finish and says successful, however, it doesn't appear to be successful.

No AP is created and the ESP32, using a serial monitor, reports nothing present to load.

April-ha, 2021-09-07:
the web flash does not transmit personal settings.
you have to use Atom.

DTTerastar, 2021-09-07:
What device are you using?

ragenhe, 2021-09-07:
ESP32 dev board.

DTTerastar, 2021-09-07:
I think I found the issue.  Deploying a fix now.  Once I test it good I'll let you know.

DTTerastar, 2021-09-07:
Ok it worked for me now.  I added the bootloader flashing this morning and hadn't fully tested it.  I needed to use the bootloader that looks for the partition table at 0x8000.  The files have a 5 minute cache so it might take that long for you to get the right one.

ragenhe, 2021-09-07:
Amazing work and turn around. Thank you!!  I will let you know what the results are as soon as I know.

ragenhe, 2021-09-07:
It is working!!!

closed 2021-09-07 by ragenhe

DTTerastar, 2021-09-07:
Awesome!