ESPresense/ESPresense #13: Web portal should load after wifi connection details established

vanstinator, 2021-09-07:
I used the website to flash an esp32 dev board. After connecting to wifi I was hoping to change additional settings from my desktop but the captive portal doesn't seem to be available. I would love to be able to hit the board at it's IP and be able to mess with additional settings.

DTTerastar, 2021-09-07:
Totally agree.  WifiSettings lib doesn't do that out of the box.  Looking for lib that supports that.

ragenhe, 2021-09-08:
It might just be a tiny web server that is needed?

DTTerastar, 2021-09-09:
PRs accepted!

czendler, 2021-09-10:
This might be useful
https://github.com/rjwats/esp8266-react

DTTerastar, 2021-09-10:
That is VERY cool.  The issue is it uses SPIFFS to store the web server contents, but we use SPIFFS to store the settings atm:

https://github.com/Juerd/ESP-WiFiSettings/blob/master/WiFiSettings.cpp#L3
https://github.com/Juerd/ESP-WiFiSettings/blob/master/WiFiSettings.cpp#L43

If we added that everyones ESPs would forget what Wifi to connect to every new firmware.

RichieFrame, 2021-09-12:
Is that a huge problem though? I have no issue reconnecting a few devices back to wifi if it means I can easily change a setting later. Later down the line could it be possible to use versioning? Can you store the web server contents AND the settings, but not have either of them be overwritten during the upgrade? That way, you could have major upgrades that overwrite SPIFFS, and minor upgrades that do not, or is that not possible?

DTTerastar, 2021-09-12:
Not unsurmountable for sure.

vanstinator, 2021-09-13:
Personally a firmware update resetting wifi settings would ensure I rarely
to never update. That's a huge hassle.

DTTerastar, 2021-09-20:
v1.5.5 allows you to change the maxDistance via mqtt.

DTTerastar, 2021-09-21:
This has been implemented in master.  Any kind souls want to test it before I roll it out?

RichieFrame, 2021-09-21:
downloading

DTTerastar, 2021-09-21:
Released!

closed 2021-09-21 by DTTerastar