ESPresense/ESPresense #7: define room

April-ha, 2021-09-07:
Hello
Congratulations on taking over this great project.
I have flashed the ESP32 and now I want to change the name of the room.
I can't find the following line
//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "living-room"

can i add it to change the name of my room?

Thanks again for this great project.

DTTerastar, 2021-09-07:
It's part of the wifi connect screen, towards the bottom.  Depending on your device you can hold down the button to clear the config and restart the wifi connect.

April-ha, 2021-09-07:
i have this https://www.mouser.fr/images/espressifsystems/lrg/ESP32-DEVKITC.jpg
and hold down the button not clear the config.

DTTerastar, 2021-09-07:
I'll add an "erase" option to the web flasher so you can use that as an alternative.  I don't have a flavor for the devkit, but I def should add it since it seems popular.  Can you see what GPIO the button is hooked up to?

DTTerastar, 2021-09-07:
I think it's GPIO15.  Try using https://github.com/ESPresense/ESPresense/releases/tag/v1.1.2 and see if it works!

DTTerastar, 2021-09-08:
Ok, there is now an option to do this: [ ] Clean install (allows you to change settings). See if it works for you.

April-ha, 2021-09-08:
Thank you for your answers.
I tried to press the "EN" button it restarts the ESP32 and the "BOOT" button does nothing.
I found this documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bootloader-num-pin-factory-reset
but the clean function of the website works well, it clears the wifi settings.

I don't see under home assistant the software version ?
https://ibb.co/hdHY6bL

DTTerastar, 2021-09-09:
BOOT is GPIO0 actually, that's not useful for clearing since it goes into bool loader mode if you press that.  Some boards just can't erase via a button.  The m5 ones all support it tho.  

Not seeing the version number means it's not an github official build.  I'll see if I broke something.

DTTerastar, 2021-09-09:
Closing this since the original issue is fixed

closed 2021-09-09 by DTTerastar