ESPresense/ESPresense #17: DOC: Home Assistant Configuration

jsBlackBelt, 2021-09-08:
Hello,

Looking at the docs found [here](https://espresense.com/home_assistant), it describes a few ways to see the serial logs:
```
...
The easiest way to find it is to connect the usb cable and monitor the serial port (115200 baud) while bringing a device close to the ESP32. To monitor the serial port you can use Ardiuno monitor function or via platform io pio run --target monitor. You should see output similar to the following:
...
```

I suggest adding info about the `PlatformIO` VScode plugin as another alternative, as it has a serial monitor as well.

DTTerastar, 2021-09-08:
Documentation is def my weakness, can you hit edit on here and PR a fix? https://github.com/ESPresense/ESPresense.github.io/blob/master/home_assistant.md

jsBlackBelt, 2021-09-09:
Yup, sure. I'll close this issue and open a new one there.

closed 2021-09-09 by jsBlackBelt