ESPresense/ESPresense #18: Devices using the same ID

feletoch, 2021-09-08:
I have a Mi Band 5 and a Amazfit Bip S. they have the same id
[image]

DTTerastar, 2021-09-08:
Are the mac's randomized?  If not we can just have it use the mac as the fingerprint for 0x0157 which is apparently Anhui Huami Information Technology Co., Ltd. (http://domoticx.com/bluetooth-company-identifiers/)

If not we can have the verbose flavor print out all the manufacture data so we can find something to add to the fingerprint (:26 just means there is 26 bytes of data).

I WISH we could use name as id, but it only comes in 3 out of 4 packets and by then we've already assigned an id.

feletoch, 2021-09-08:
I restarted the watch and they seem to use the same mac afterward

DTTerastar, 2021-09-09:
v1.2.0 should resolve the issue, i'm using the mac as the fingerprint for that case now.

feletoch, 2021-09-13:
That's awkward, the issue is back
[image]

DTTerastar, 2021-09-13:
Ok, I know what that is... Multiple advertisements w/ the same mac causes issues.  Same as #23

DTTerastar, 2021-09-15:
Ok, the newest build should fix this.  It might for 1 packet report as sid:* but once the md packet is received it'll use the mifit:<mac> as the id. 

I found this interesting doc about that device: http://changy-.github.io/articles/xiao-mi-band-protocol-analyze.html

closed 2021-09-15 by feletoch