ESPresense/ESPresense #20: All Sonos devices show the same id "md:05a7:150"

Scott8586, 2021-09-10:
**Describe the bug**
All Sonos devices show the same md based id, including the portable.

**Logs**

```
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBCA933ACE9","rssi@1m":-41,"rssi":-76,"mac":"d2d26988f1e1","raw":10,"distance":11.05,"speed":-0.3}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"846B210D346030AEE9","rssi@1m":-41,"rssi":-84,"mac":"f333d57f21aa","raw":16.93,"distance":13.72,"speed":4.6}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBCA933ACE9","rssi@1m":-41,"rssi":-78,"mac":"d2d26988f1e1","raw":11.41,"distance":11.07,"speed":0.2}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBC3433ACE9","rssi@1m":-41,"rssi":-81,"mac":"d2d24cc6da8a","raw":13.89,"distance":15.76,"speed":-0.5}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"846B210D346030AEE9","rssi@1m":-41,"rssi":-82,"mac":"f333d57f21aa","raw":14.84,"distance":14.48,"speed":0.4}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBCA933ACE9","rssi@1m":-41,"rssi":-77,"mac":"d2d26988f1e1","raw":10.68,"distance":11.29,"speed":-0.2}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBC3433ACE9","rssi@1m":-41,"rssi":-80,"mac":"d2d24cc6da8a","raw":13.01,"distance":15.54,"speed":-0.3}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"846B210D346030AEE9","rssi@1m":-41,"rssi":-77,"mac":"f333d57f21aa","raw":10.68,"distance":14.07,"speed":-3.3}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBCA933ACE9","rssi@1m":-41,"rssi":-76,"mac":"d2d26988f1e1","raw":10,"distance":11.12,"speed":-0.8}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"F8B3C5DDBC3433ACE9","rssi@1m":-41,"rssi":-78,"mac":"d2d24cc6da8a","raw":11.41,"distance":14.96,"speed":-1.4}
espresense/rooms/familyroom {"id":"md:05a7:150","name":"846B210D346030AEE9","rssi@1m":-41,"rssi":-83,"mac":"f333d57f21aa","raw":15.85,"distance":14.16,"speed":0.7}
```
I think this will probably be fixed if I upgrade to 1.2.0. but thought this would be informational.

Scott8586, 2021-09-10:
Correction, I'm already on 1.2.0 - so this is a bona fide bug.

DTTerastar, 2021-09-10:
Have you looked through the serial log to try and find something to use to keep them separate?  It looks like mac's are randomized so we can't use that.  md:05a7:150 means there is 150 bytes of stuff being sent by the sig assigned company "Sonos": https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/

Compare the serial output and find a part that is static and we can use that for the fingerprint.

Scott8586, 2021-09-10:
The MAC addresses are not randomized, they are fixed - look at the original report, you'll see the MAC addresses are repeating (that report has three Sonos devices represented). And they use that fixed address across reboots.

DTTerastar, 2021-09-10:
Hmmm, ok, then they must be a different flavor than "Public":

https://github.com/ESPresense/ESPresense/blob/e261f715725fc948ab0b090f1a1d5c636aa17c6d/lib/BleFingerprint/BleFingerprint.cpp#L11

Any chance you can add a serial log to that line and flash it to find out?

sourbrambles, 2021-09-11:
For Info, I'm seeing exactly the same experience with Google Home Minis.

Scott8586, 2021-09-11:
FWIW, I looked with one of the other BLE room tools, "monitor" and the MAC address appears to be advertised as random, but at the practical level the MAC address is fixed, I've never seen them change.  I'll try to get some more data off a serial connection.

DTTerastar, 2021-09-11:
Lot's of devices just won't have unique fingerprints.  It's more useful to have a fingerprint that you can say that a device is in a room, instead of just seeing lot's of random macs that are useless.  The real trick is finding something you can use to make a unique fingerprint.  With apple I found that the first 4 hex chars never changed, and were different based on model/ios version so I added that to the fingerprint so I could tell the difference between my watch and my wife's.

DTTerastar, 2021-09-11:
> FWIW, I looked with one of the other BLE room tools, "monitor" and the MAC address appears to be advertised as random, but at the practical level the MAC address is fixed, I've never seen them change. I'll try to get some more data off a serial connection.

Hmmm, that is wierd.  It just may take longer for it to randomize.  Apple is crazy fast at 15m.

DTTerastar, 2021-09-11:
The other thing I could add would be to use the "Name" that it's broadcasting.  We'd use the id until it finds the name, and then from then it could use that until the mac changes (which isn't very often anyways).

Scott8586, 2021-09-11:
The name field does appear to be constant.  When I say I haven't seen the MAC change, I mean - I've had these devices 18months, and the MAC addresses still match my initial scans.

DTTerastar, 2021-09-12:
Try v1.2.0 as see if it fixes this for sonos.

DTTerastar, 2021-09-15:
v1.4.0 has a fix for this, reopen this if it's not fixed.

Scott8586, 2021-09-15:
Not fixed,  all Sonos devices show as sid:0xfe070

DTTerastar, 2021-09-16:
1.4.4 has more checks for sonos, let me know if it fixes it.

Scott8586, 2021-09-16:
Fixed.

commit a4fe08d8 (2021-09-15): Fixes #20 and #23
closed 2021-09-16 by Scott8586