ESPresense/ESPresense #22: WiFi pins to BSSID, not SSID, which can result in poor connectivity

anaisbetts, 2021-09-10:
**Describe the bug**

When configuring Wifi, ESPresence saves the MAC address of the AP rather than the SSID. This means that in a Mesh network / multi-AP house, it's very possible to pick the wrong AP, resulting in devices having super poor connectivity despite being near an AP

**To Reproduce**
1. Go to an area with more than one AP with the same SSID
2. Pick the SSID you think is correct, but oops it's not
3. Restart the device and check the reported RSSI, see that it is very poor

**Expected behavior**
The device should save SSIDs not MAC Addresses, then on boot do a scan and pick the AP that has the highest signal strength

DTTerastar, 2021-09-11:
Can you report that to https://github.com/Juerd/ESP-WiFiSettings?  That's what handles that.

DTTerastar, 2021-09-20:
I'm closing since I'm pretty sure we only use SSID to connect...

closed 2021-09-20 by DTTerastar