ESPresense/ESPresense #24: Multiple Devices show id as exp:20

skycryer, 2021-09-12:
**Describe the bug**
Multiple Devices show ID: exp:20 as identifier. I have an iPhone 7 and 11 here. The iPhone 11 shows as apple:1007:11-7 the iPhone 7 as exp:20.

**Expected behavior**
iPhone 7 should also have a unique id

**Logs**
0 New   | MAC: 558c95b262fa, ID: apple:1006:10-6
0 New   | MAC: 565877bd5177, ID: apple:1006:10
0 New   | MAC: 6d0d3a32b91a, ID: apple:1006:10
0 New   | MAC: 6ba18edc5499, ID: apple:1007:11-7
1 Close | MAC: 6d0d3a32b91a, ID: apple:1006:10                                     
0 New   | MAC: c0288d341827, sID: 0xfe61
0 New   | MAC: 6c7b4ef1fe79, ID: apple:1005:9-26
1 Close | MAC: 3eb56c68d2ce, ID: exp:20                                            
0 New   | MAC: a4c138a6748f, ID: a4c138a6748f, MD: 0188ec000101
1 Close | MAC: 6ba18edc5499, ID: apple:1007:11-7                                   
0 New   | MAC: 7b29ac2c42b3, ID: apple:1005:9-12
1 Close | MAC: 1e399b08ae07, ID: exp:20                                            
1 Left  | MAC: 6d0d3a32b91a, ID: apple:1006:10                                     
0 New   | MAC: 707c6a4d3c3b, ID: apple:1006:10-6
1 Left  | MAC: 6ba18edc5499, ID: apple:1007:11-7                                   
1 Left  | MAC: 1e399b08ae07, ID: exp:20                                            
1 Close | MAC: 1e399b08ae07, ID: exp:20                                            
1 Close | MAC: 6ba18edc5499, ID: apple:1007:11-7                                   
0 New   | MAC: 4a7575ed2e88, ID: apple:1005:9-12
0 New   | MAC: 3414a6a085ba, ID: exp:20
1 Left  | MAC: 6ba18edc5499, ID: apple:1007:11-7                                   
1 Left  | MAC: 1e399b08ae07, ID: exp:20                                            
0 New   | MAC: 680ae219cf13, ID: 680ae219cf13, MD: e70601030c030102001c5c88bfe7d2094b5d881ef32f89eeb403
1 Close | MAC: 3414a6a085ba, ID: exp:20                                            
1 Left  | MAC: 3414a6a085ba, ID: exp:20                                            
1 Close | MAC: 3414a6a085ba, ID: exp:20                                            
1 Left  | MAC: 3414a6a085ba, ID: exp:20

DTTerastar, 2021-09-12:
exp:20 is covid exposure tracking.  the teks they send are not trackable individually.  If you want to use exp for iPhone monitoring you need to uninstall the exposure tracking on the other devices.

skycryer, 2021-09-12:
Good to know, but why is this the only id my iphone7 is generating? I tried it several times putting it near and far away but the only id I can see is the exp:20.

DTTerastar, 2021-09-12:
The nearby info is broadcast to support handoff, air drop, etc.  Try turning on those features if they aren't already on.  This is totally undocumented stuff so I'm not really sure.  I've added some details to here: https://espresense.com/beacons/apps#ios--watch-os-continuity-fingerprints

skycryer, 2021-09-13:
Tried several thing like activate nearby actions, check bluetooth, data protection, location services but I can not get the iPhone to be seen as the other devices here do. Any other idea? As far as I saw I do not need an app as on android, correct? I have iOS 14.4.7 on both iPhones but only one is visible.

DTTerastar, 2021-09-13:
> Tried several thing like activate nearby actions, check bluetooth, data protection, location services but I can not get the iPhone to be seen as the other devices here do. Any other idea? As far as [...]

The only app that Apple allows to broadcast btle on the iphone is the covid exposure.  So uninstall it from the other devices and use it on the 7.  Problem solved.

closed 2021-09-13 by DTTerastar

sblessley, 2023-01-06:
@DTTerastar is there a list of these codes (e.g.exp:20) somewhere? I didn't see any on the ESPresence website.