ESPresense/ESPresense #25: Show names of device for identifiy the correct device

skycryer, 2021-09-12:
**Is your feature request related to a problem? Please describe.**
When connecting via Arduino tool you can only see the Mac address and the generated id but not the name of a device if it is sending a name

**Describe the solution you'd like**
Show Device names where available like this tool does https://github.com/jnross/Bluetility

DTTerastar, 2021-09-12:
That tool is very nice and I can see quite a few more names.  That would be great if we could get all that.

DTTerastar, 2021-09-15:
I've been able to get a few more names, is it any better for you?

skycryer, 2021-09-16:
I get some names but the apple devices all do not show names as the do in the app above

DTTerastar, 2021-09-16:
I think that's because of icloud magic that adds btle connections for you.  I'm not sure we can do that without a lot of reverse engineering.  If you could see those names on a mac that you haven't logged into the same icloud that would mean it was possible.

skycryer, 2021-09-16:
Hm, would need to check that with a Mac not logged in

DTTerastar, 2021-10-05:
When `Query` is on in v2.0.0 ESPresense can get the name of my MacBook Pro.  I only have enabled query for apple devices atm.  Is there a specific device you need the name for?

skycryer, 2021-10-05:
Names of iPhones or apple devices in general would be good, would make it easier.

DTTerastar, 2021-10-05:
I haven't seen iPhones give up that info when not securely paired.  They do give up the model tho.  v2.0.0 supports the room assistant companion app to assign a unique id.

skycryer, 2021-10-05:
Is there a docu how to do that?

DTTerastar, 2021-10-05:
Just install the latest and check this box:
[image]

closed 2021-10-11 by DTTerastar