ESPresense/ESPresense #26: Not connecting to MQTT

brewtangclan, 2021-09-12:
**Describe the bug**
I have flashed two ESP32s using the browser-based installer, and after setting Wifi/MQTT settings in the captive portal, the ESP32s connect to the WiFi network but appear to never connect to MQTT (as evidenced by serial logs, Mosquitto logs, and lack of espresence MQTT topics). This is occurring both with the default ESP32 build and the verbose logging build.

**Expected behavior**
ESP32 connects to WiFi, and then MQTT

**Logs**
Serial log:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi ([MAC redacted])...
I (65) wifi:wifi driver task: 3ffd207c, prio:23, stack:3584, core=0
I (257) wifi:wifi firmware version: dc30037
I (258) wifi:config NVS flash: enabled
I (258) wifi:config nano formating: disabled
I (258) wifi:Init data frame dynamic rx buffer num: 32
I (263) wifi:Init management frame dynamic rx buffer num: 32
I (268) wifi:Init management short buffer num: 32
I (273) wifi:Init dynamic tx buffer num: 32
I (277) wifi:Init static rx buffer size: 1600
I (281) wifi:Init static rx buffer num: 16
I (284) wifi:Init dynamic rx buffer num: 32
I (385) wifi:mode : sta (24:62:ab:ca:f9:e4)
Connecting to WiFi SSID [SSID redacted].....I (2454) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2455) wifi:state: init -> auth (b0)
I (2462) wifi:state: auth -> assoc (0)
I (2470) wifi:state: assoc -> run (10)
I (2501) wifi:connected with [SSID redacted], aid = 77, channel 11, BW20, bssid = [BSSID redacted]
I (2502) wifi:security type: 3, phy: bgn, rssi: -45
I (2512) wifi:pm start, type: 1

I (2513) wifi:AP's beacon interval = 102400 us, DTIM period = 1
10.x.x.116
Version:     v1.3.0
IP address:  10.x.x.116
DNS address: 10.x.x.1
Hostname:    espresense-test
Room:        test
Telemetry:   enabled
Rooms:       enabled
Devices:     enabled
Max Distance: 16
```
Mosquitto logs: No record of "New client connected from 10.x.x.116" 
MQTT topics: No espresence topic exists

**Hardware Details**
This is occurring for me on both a Firebeetle ESP32 as well as the "standard" ESP32 devkit
MQTT is a fairly standard Mosquitto implementation, no SSL, listening on the same LAN as the ESP32 (I have other ESP32s deployed with Tasmota/Arduino around the house without issues). 
Install was done using the latest version of Chrome. Also tested using Edge with identical results.

Captive Portal Settings:
[image]

DTTerastar, 2021-09-12:
It's stuck trying to sync time:

```void setClock()
{
    configTime(0, 0, "pool.ntp.org", "time.nist.gov"); // UTC

    time_t now = time(nullptr);
    while (now < 8 * 3600 * 2)
    {
        yield();
        delay(500);
        now = time(nullptr);
    }

    struct tm timeinfo;
    gmtime_r(&now, &timeinfo);
    log_i(F("NTP synced, current time: %s"), asctime(&timeinfo));
}```

Do you restrict it from the internet by chance?

DTTerastar, 2021-09-12:
I think I can actually remove this.  I ended up not actually needing the time.

brewtangclan, 2021-09-12:
> Do you restrict it from the internet by chance?

I sure do, my IOT subnet for ESPs does not get internet access - that would certainly explain it. My MQTT server also happens to be my local SNTP server - could NTP settings be parameterized / added to the captive portal?

Scott8586, 2021-09-12:
I would like to control the NTP sever too, can you take it from DHCP, please?

DTTerastar, 2021-09-13:
I didn't actually end up using the time for anything so I've just disabled it for now:

https://github.com/ESPresense/ESPresense/releases/tag/v1.3.1

Please try that.

brewtangclan, 2021-09-13:
Can confirm - v1.3.1 connects to MQTT and detects devices without internet access.

closed 2021-09-13 by DTTerastar