ESPresense/ESPresense #31: AppDaemon install issues.

michaeldvinci, 2021-09-14:
**Describe the bug**
Chatted with you previously. Manual install [for me] across all version results in the following error: 'NoneType has no mqtt.unsubscribe'

Also, can't add appdaemon github repo as HACS repo.

**Expected behavior**
appdaemon app runs.

**Logs**
 'NoneType has no mqtt.unsubscribe'

**Hardware Details**
only esp32 dev kit. do we need gps modules to work with the appdaemon?

**Screenshots**
n/a

mvinci-wyze, 2021-09-14:
relevant screenshots:

[appdaemon logs](https://i.imgur.com/ggT3AA0.png)
[appdaemon app folder structure](https://i.imgur.com/FFrqLtf.png)

michaeldvinci, 2021-09-14:
relevant screenshots:

[appdaemon app folder structure](https://i.imgur.com/FFrqLtf.png)
* The screenshot/logs below seem to be referring to `self.mqtt = self.get_plugin_api("MQTT")` on line 25 in this screenshot

[appdaemon logs](https://i.imgur.com/ggT3AA0.png)

```
2021-09-14 20:38:24.313166 INFO AppDaemon: Loading App Module: /config/appdaemon/apps/espresense-ips.py
2021-09-14 20:38:24.810126 INFO AppDaemon: Initializing app hello_world using class HelloWorld from module hello
2021-09-14 20:38:24.814989 INFO AppDaemon: Initializing app espresense using class ESPresenseIps from module espresense-ips
2021-09-14 20:38:25.117476 INFO hello_world: Hello from AppDaemon
2021-09-14 20:38:25.121156 INFO hello_world: You are now ready to run Apps!
2021-09-14 20:38:25.127485 WARNING AppDaemon: Unknown Plugin Configuration in get_plugin_api()
2021-09-14 20:38:25.132450 INFO espresense: Subscribing to topic espresense/rooms/bedroom
2021-09-14 20:38:25.135803 WARNING espresense: ------------------------------------------------------------
2021-09-14 20:38:25.136790 WARNING espresense: Unexpected error running initialize() for espresense
2021-09-14 20:38:25.137643 WARNING espresense: ------------------------------------------------------------
2021-09-14 20:38:25.142868 WARNING espresense: Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/appdaemon/app_management.py", line 165, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.9/site-packages/appdaemon/utils.py", line 308, in run_in_executor
    response = future.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/appdaemon/apps/espresense-ips.py", line 29, in initialize
    self.mqtt.mqtt_unsubscribe(t)
AttributeError: 'NoneType' object has no attribute 'mqtt_unsubscribe'
2021-09-14 20:38:25.143776 WARNING espresense: ------------------------------------------------------------
```

DTTerastar, 2021-09-15:
Ahh, That's because you need to hook appdaemon up to both hass and mqtt:
```
  plugins:
    HASS:
      type: hass
      ha_url: "redacted"
      token: "redacted"
      namespace: default
      app_init_delay: 30
      appdaemon_startup_conditions:
        delay: 30
    MQTT:
      type: mqtt
      client_host: "redacted"
      namespace: mqtt
      birth_topic: appdaemon
      will_topic: appdaemon
```