ESPresense/ESPresense #33: Home Assistant Android App BLE beacon not showing up in MQTT messages

mloebl, 2021-09-14:
**Describe the bug**
I've enabled the BLE Beacon in the Android app, I see the iBeacon messages coming into the terminal, but not in MQTT Explorer. I am seeing other devices fine.

Not sure if related,. but they are also strangely backwards... In the HA Android app it says "112233445566778899AABB-100-1" however in the serial log it shows "BBAA998877665544332211-100-1" (it's actually a longer UUID string, but hopefully you get the idea, note they are reversed based in pairs every time, not in full reverse order, if a separate bug and not related, I can open one up.)  

Neither of these strings are showing up in the MQTT app when watching with MQTT Explorer. 

**Expected behavior**
The UUID for the ID should show up in MQTT and match the expected one in the app

**Logs**
```
0 New   | MAC: 7cd1b850ed1d, ID: iBeacon:REMOVED-100-1
1 Del   | MAC: 63e2f09760bc, ID: iBeacon:REMOVED-100-1
```

**Hardware Details**
Android Pixel 4XL running app version beta-731-c5e3d4a-full
ESP32 D1 Mini

**Screenshots**

mloebl, 2021-09-15:
I just checked with 1.4.0, still not change, the HA iBeacons don't show up in MQTT but are showing up in the serial log

DTTerastar, 2021-09-15:
I need to locate a android phone to test... I had one but my wife washed it...

mloebl, 2021-09-15:
> I need to locate a android phone to test... I had one but my wife washed it...

haha been there before :)  Is there a way I can help debug in the mean time?

DTTerastar, 2021-09-15:
Can you try the other 2 beacon apps and see if they work, or are similarly broke?

mloebl, 2021-09-15:
BeaconScope, while backwards in both serial output and MQTT like the HA UUID being backwards, it did show up in MQTT. So something in the message from HA it may not be liking? I tried 2 different HA Android phones we have, neither seem to work so it's app specific. I just realized I do not have the Verbose logging version installed, let me do that and see what I can see.

mloebl, 2021-09-15:
Not seeing anything different other than a mDNS failure on startup with verbose, there's nothing showing when the messages are received. I may be able to fire up platformio later to see if I can debug a bit as well if work isn't too crazy.

mloebl, 2021-09-15:
@DTTerastar Not sure if it's a coincidence yet or not, but I updated the platformio dependency of NimBLE-Arduino from 1.20 to h2zero/NimBLE-Arduino@^1.3.1 and seems to be working now. Watching it to monitor to see if fixed or not, but it's now showing up.

mloebl, 2021-09-15:
Ugh it's a coincidence, it's working intermittently; as in showing up consistently in the serial output but not in MQTT.

DTTerastar, 2021-09-15:
I just pushed a fix that may help.  It fixed a similar issue w/ keysmart
devices.  It was because they were sending multiple uuids in one broadcast.

mloebl, 2021-09-15:
Just updated, but starting to wonder if something isn't quite right with HA's app or whatever it's sending. I switch to the BeaconScope app and it works solid, however was hoping to not have two apps for the same thing.

DTTerastar, 2021-09-16:
I figured it out.  It broadcasting by default at VERY low power, but i'm still reading that is should be -59 rssi@1m.  At 0.1m i'm only seeing -74.  So the reason you aren't seeing it in mqtt is it's being detected as greater than maxDistance away which stops the send to mqtt.

DTTerastar, 2021-09-16:
You are right about the iBeacon being backwards.  I think that is an issue in the ha companion app.  We also need them to send an accurate calibrated rss@1m.  That should also should be editable so you can actually calibrate it.

I just set the beacon to full power and and I now see it in all at least 3 rooms.  So I don't believe there is any issue in ESPresense.

DTTerastar, 2021-09-16:
https://github.com/home-assistant/android/blob/fe8aa454de453f9a079655f0500e6cf585916e62/app/src/main/java/io/homeassistant/companion/android/bluetooth/ble/TransmitterManager.kt#L18 See how it's -59 no matter what power level you choose?  That's gotta be fixed.

DTTerastar, 2021-09-16:
Added this: https://github.com/home-assistant/android/issues/1715

mloebl, 2021-09-16:
Thank you very much for digging into this!

mloebl, 2021-09-16:
I also suspect this is why when I was checking out the new ESPHome feature released with BLE it didn't work either and never saw the phone...

DTTerastar, 2021-09-16:
The UUID I was reading was backwards.  I cross checked w/ BlueCharm.  Fixed in v1.5.0

mloebl, 2021-09-16:
> The UUID I was reading was backwards. I cross checked w/ BlueCharm. Fixed in v1.5.0

Confirmed! Thank you for that :)

DTTerastar, 2021-09-17:
Good to close this?

mloebl, 2021-09-17:
Yup! Stuck waiting for the Android app issue to be looked at this point. Thank you for the help!

closed 2021-09-17 by mloebl