ESPresense/ESPresense #34: Device won't connect to wifi resets every time its rebooted.

bluerhino4, 2021-09-14:
Flashing the esp32 is successful and it loads the AP to connect to for initial setup. (wifi, MQTT) When you click save the web page reloads with the wifi password censored, but it does not connect to the network. Clicking restart device just restarts it and it loads back up as if it had just been flashed. I have tried with two different wifi networks as one is a mesh network and somethings just don't like it. 

The only thing that I could see being the problem is that I am using an ESP32S CAM dev board, and I'm not sure if espresence is supported. (Waiting on different ESP32s just using CAM to test in the mean time)
I have tried two different ESP32S Cam boards. 

There is one error that pops up in my serial monitor which seems to be related to the webserver so I have been assuming that it is not the cause of my problems. 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮Resetting back to defaults...
Connecting to WiFi (CC:50:E3:94:ED:18)...
First contact!

Starting access point for configuration portal.
SSID: 'espresense-18ed94'
192.168.4.1
[E][WebServer.cpp:633] _handleRequest(): request handler not found
[E][WebServer.cpp:633] _handleRequest(): request handler not found
6 WiFi networks found.
[E][WebServer.cpp:633] _handleRequest(): request handler not found
[E][WebServer.cpp:633] _handleRequest(): request handler not found
6 WiFi networks found.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Resetting back to defaults...
Connecting to WiFi (CC:50:E3:94:ED:18)...
First contact!

Starting access point for configuration portal.
SSID: 'espresense-18ed94'
192.168.4.1

lmanoarchilla, 2021-09-15:
I have the same problem with sparkfun esp32 thing, i disconnect it from PC a reconect with another wire and flash it and it begins to work

DTTerastar, 2021-09-15:
I don't have a ESP32S2 specific build atm, but I can try and add one.

bluerhino4, 2021-09-15:
That would be awesome but personally speaking I was only planning on using this module to test while I wait for my ESP32 WROOMs to arrive. If you want to/have time to make a specific build for ESP32S then I will happily test it out but I can also just wait for the supported ESP32.

DTTerastar, 2021-09-15:
Ok, the issue is it "thinks" the reset settings button is down.  I'll make a flavor that just doesn't care about that.

DTTerastar, 2021-09-16:
Ok, I just made 1.4.4 not do any button handling for esp32, and now it works on my esp32 cam.

bluerhino4, 2021-09-16:
Prefect thank you so much, I really appreciate it. I have been able to successfully flash and configure my esp32 and it seems to be working. Just need to get it into HA now.

closed 2021-09-16 by bluerhino4