ESPresense/ESPresense #35: Signature of iOS device changes after OS upgrade

Scott8586, 2021-09-15:
**Describe the bug**
Signature changes after OS upgrade

**Expected behavior**
Not certain, signature remain constant.

**Hardware Details**
Two iPhone 12 Pro updated to 14.8

Espresense 1.2.0

Scott8586, 2021-09-15:
In fact, signature Id changes every time I reboot or toggle Bluetooth.

Scott8586, 2021-09-15:
I can confirm that an id is stable on an iPad running iPadOS 14.7.1, after Bluetooth toggle.

DTTerastar, 2021-09-15:
What id's are you seeing?  Only the apple:10xx-* ones are typically useful.  07 is handoff and will come and go based on using your device.

Scott8586, 2021-09-15:
So far, for the same device ive seen:

1006:10-6
1006:10-8
1002:6-6

After multiple reboots, etc, it seems to be stabilizing on 1006:10-6. Before this update it was 1005:9-6

Why would volts be reported as an attribute?

DTTerastar, 2021-09-15:
I've added Eddystone TLM which does have voltage reported.  But I think I have an uninitalized variable or something since I doubt iphone reports that.

Scott8586, 2021-09-15:
OK, this is part of what was confusing me. Now both our iPhones are showing up 1006:10-6, I can see the Mac address flip back and forth.  Here they are in two separate rooms up each near the ESP32:

```
espresense/devices/apple:1006:10-6/retreat {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-44,"mac":"79cf95716695","raw":0.82,"distance":0.87,"speed":0,"volts":58143}
espresense/devices/apple:1006:10-6/retreat {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-44,"mac":"79cf95716695","raw":0.82,"distance":0.87,"speed":0,"volts":58143}
espresense/devices/apple:1006:10-6/familyroom {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-34,"mac":"40aec444fd92","raw":0.43,"distance":0.53,"speed":0,"volts":25568}
espresense/devices/apple:1006:10-6/retreat {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-46,"mac":"79cf95716695","raw":0.94,"distance":0.86,"speed":0,"volts":58143}
espresense/devices/apple:1006:10-6/familyroom {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-34,"mac":"40aec444fd92","raw":0.43,"distance":0.52,"speed":0,"volts":25568}
espresense/devices/apple:1006:10-6/retreat {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-46,"mac":"79cf95716695","raw":0.94,"distance":0.87,"speed":0,"volts":58143}
espresense/devices/apple:1006:10-6/familyroom {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-34,"mac":"40aec444fd92","raw":0.43,"distance":0.51,"speed":0,"volts":25568}
espresense/devices/apple:1006:10-6/retreat {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-46,"mac":"79cf95716695","raw":0.94,"distance":0.88,"speed":0,"volts":58143}
espresense/devices/apple:1006:10-6/familyroom {"id":"apple:1006:10-6","rssi@1m":-47,"rssi":-34,"mac":"40aec444fd92","raw":0.43,"distance":0.5,"speed":0,"volts":25568}
^
```

DTTerastar, 2021-09-15:
Yeah, if the the model and ios are very similar you may not be able to tell them apart.  That's just a limitation of the fingerprinting.  I do want to make an appdaemon app that'll "count" the number of devices in a room, so you could make an automation for when 0 iphones in a room as well as another automation for when number of iphones is >0.

closed 2021-09-15 by DTTerastar