ESPresense/ESPresense #39: Generated iPhone IDs not unique to each iPhone

RutgerDiehard, 2021-09-16:
I have 5 iPhones that I am tracking across 8 x ATOM Lite ESP32 running Espresense - all are different iPhone models and all run iOS 14.7/8:

iPhone 8 Plus, XS Max, 11 Pro Max, 12, and 12 Pro Max.

The last three of the above list work fine. However, the 8 Plus and XS Max despite having different IDs configured in Configuration.yml, their entities update with the same or slightly different status. For example, the XS Max is not physically present and cannot be seen by any ESP32 devices, however its entity location updates with the 8 Plus entity. 

I've removed the 8 Plus entity (by #ing the entry in configuration.yml) and the XS Max entity then behaves correctly. Adding it back then reproduces the problem.

DTTerastar, 2021-09-17:
I include in the fingerprint everything I know to not be random.  If you get the same fingerprint then it's either untrackable, or we need to look in more depth to find something else to add to the fingerprint.  With apple as focuses on privacy as they are, we are lucky, we can deduce this much info from the advertisements.

DTTerastar, 2021-09-19:
https://sciendo.com/article/10.2478/popets-2019-0057

This mentions we can connect to the GATT service and ask for the model string.  That would help you right?

RutgerDiehard, 2021-09-20:
> https://sciendo.com/article/10.2478/popets-2019-0057
> 
> This mentions we can connect to the GATT service and ask for the model string. That would help you right?

Yes, this looks to be the solution. Great find! Is this easy to add? I wonder if there would be a small performance hit with the GATT query.

DTTerastar, 2021-09-20:
Well I have 7 esp 32s￼, if every single one of them made this query it would be a mess￼. So I think we're in distributed computing territory now.  The ESP's need to talk to each other and have only one of them do the query...￼￼

brianlaucis, 2021-09-22:
Same issue two separate Iphone's with the name fingerprint ID.

ee02217, 2021-09-29:
I think I have the same issue. As you can see in the image below, on the left side, the person with the phone, was away from home (very very far away, and I live in a 5th floor, so signal was not captured from the street), but it still received some location updates (incorrectly):
[image]

Tokn59, 2021-09-30:
Sanme issue with the iPhone ID's .. It does show different MAC addresses in the Terminal though .. couldn't we use the MAC address iso the Apple ID:

DTTerastar, 2021-10-01:
If you watch in the terminal after 15 minutes all your apple devices will change their mac address.  I am working on supporting the home assistant app that adds an id to your iphone: https://apps.apple.com/us/app/room-assistant/id1538642237?ign-itsct=apps_box_link&ign-itscg=30200

RutgerDiehard, 2021-10-01:
> If you watch in the terminal after 15 minutes all your apple devices will change their mac address. I am working on supporting the home assistant app that adds an id to your iphone: https://apps.apple.com/us/app/room-assistant/id1538642237?ign-itsct=apps_box_link&ign-itscg=30200

That's great news @DTTerastar! I'm quite keen to work around this issue as I thought I could fix it by simply getting rid of the iPhone 8 Plus (I've just got an iPhone 13 Pro Max, so my daughter will have my old 11 Pro Max and the 8 Plus would be the "emergency phone"). Unfortunately, it seems Apple fully backs up and restores ALL configuration onto a new phone - including the elements that ESPresense uses to make up the Bluetooth ID. So now my new iPhone 13 Pro Max has the same ID as the 11 Pro Max. Million dollar question is, do you have an ETA for Room-Assistant app support?

corgan2222, 2021-10-01:
> 
> 
> If you watch in the terminal after 15 minutes all your apple devices will change their mac address. I am working on supporting the home assistant app that adds an id to your iphone: https://apps.apple.com/us/app/room-assistant/id1538642237?ign-itsct=apps_box_link&ign-itscg=30200

That's fantastic!

martinrosenauer, 2021-10-02:
Great project, thanks! 

Wondering, wouldn’t it be possible to take the same approach with Apple devices as Andrew Freyer’s monitor.sh script does ? (It works well).. https://github.com/andrewjfreyer/monitor

DTTerastar, 2021-10-03:
I think that uses classic bluetooth.  With BTLE I can get 3 position updates a second for certain devices, classic bluetooth is much slower and uses more battery.  Monitor is very clever about how often it probes for device.  The ESP32 also isn't great at classic bluetooth in general, but it is a possibility for hard to fingerprint devices...

DTTerastar, 2021-10-05:
2.0.0 is out with this feature.  You must turn on Query for it to happen, but it's not 100% reliable yet.

ivanvach, 2021-10-07:
Tested O.K. with 2 iPhones. Somewhat weird numbers but I believe it is beyond your control

iPhone12 - reported as apple:1006:10-8-iphone13-2
iPhone11 - reported as apple:1006:10-8-iphone12-3

Without the Query active they both would be reported as 

 apple:1006:10-8

obviously you did a very good job. Thanks.

DTTerastar, 2021-10-09:
Turn on the "Query" feature to ask the phones what model number they are to see if that makes the fingerprint unique.

Del is the opposite of New.  It means we haven't seen this mac for 300 seconds so were stopping tracking it.  Once it comes back it'll be a New.  We keep a bunch of filters (to reduce rssi noise) per mac so we can't keep all macs in memory.

Tokn59, 2021-10-10:
This is the result :

[image]

Tokn59, 2021-10-10:
Some of my devices are Apple Airtags. Interesting article about changing their ID's can be found here : https://adamcatley.com/AirTag

DTTerastar, 2021-10-11:
Install https://apps.apple.com/us/app/room-assistant/id1538642237?ign-itsct=apps_box_link&ign-itscg=30200 you'll then see the uuid shown in the app.

Tokn59, 2021-10-12:
Hi, I successfully installed the room-assistant integration in Home Assistant via MQTT. First conclusion is that indeed the iPad Mini does not update regularly (has been doing it once or twice) and that the distance reported for the iPhone is far off real distance so I should use a "compensation index" here. For instance when it indicates 3.28 meters as in example here it should be at least 11-12 meters through air.

[image]

DTTerastar, 2021-10-12:
I've never used the room assistant integration.  I haven't designed things to be compatible.  Just the app so far.

DTTerastar, 2021-10-12:
Do you mean mqtt_room?

Tokn59, 2021-10-12:
> Do you mean mqtt_room?

I use both now in Home Assistant. mqtt_room for the Apple Airtags (not working well, ID's change constantly) and the normal sensor MQTT for the roomassistant MQTT messages

DTTerastar, 2021-10-12:
I'm going to close this since it's largely fixed with the room assistant app support, but specific issues w/ getting the new fingerprints to work can be opened as separate issues.