ESPresense/ESPresense #40: Manual installation docs?

phrozen77, 2021-09-16:
I wonder, could we get manual installation instructions somewhere?

I just stumbled upon this project and would like to give it a spin, been searching for a solution like this for ages (can't dedicate multiple RasPis to the task etc.etc...)

So, my problem is that i neither run Google Chrome nor Microsoft Edge but could flash an ESP32 with esptool or similar...

Thank you for your consideration & your hard work!

dcgrove, 2021-09-16:
Download the bin file from the release page and flash it with your preferred flashing tool. That is how I flashed mine.

DTTerastar, 2021-09-16:
VSCode w/ the Platformio plugin is how I do it.
You just click the Upload and Monitor

This should work as well: https://github.com/esphome/esphome-flasher/releases

closed 2021-09-17 by DTTerastar

dattasaurabh82, 2022-02-22:
**Question:** How to install the [esp32 bin file](https://github.com/ESPresense/ESPresense/releases/download/v2.0.31/esp32.bin) using `esptool.py`? Any special flags?

I'm using this to erase and flash:
```shell
sudo esptool.py --port /dev/tty.usbserial-016420E5 --baud 115200 —chip esp32 erase_flash
sudo esptool.py --port /dev/tty.usbserial-016420E5 --baud 115200 —chip esp32 write_flash 0x000000 esp32.bin
```

Flashes successfully:
```shell
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:14:9f:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0017afff...
Compressed 1551952 bytes to 907348...
Wrote 1551952 bytes (907348 compressed) at 0x00000000 in 79.7 seconds (effective 155.7 kbit/s)...
Hash of data verified.
```
But no AP after that. :[

Where as if I do it using the esp web flasher (from the install page), then everything works. 

My HW is Adafruit Huzzah 32 and ESP-WROOM-32 and both of them works, when installed from webpage.

_Anything missing from the esptool.py flags?_

ctschach, 2022-02-22:
I run into the same issue. The problem is, that the ESP.bin does not include the full package including Bootloader and Partition table. 

I ended up installing "ESPHome-Flasher" as described in the installation instruction on my Mac and flashed my first ESP32 that way. Then I took a full 4MB dump with esptool.py on my raspberry and now I have an image which I can flash to all new ESP32s.

dattasaurabh82, 2022-02-22:
@ctschach So basically extract the pre-installed binary(the one flashed through either webUI or other GUI tools) in 4 mb file size, then use `esptool.py` to use the full mem (4mb partition) to re_upload? 
Will give it a try tomorrow and report back

ctschach, 2022-02-22:
I found this webpage helpful:

https://community.blynk.cc/t/how-to-backup-restore-official-firmware-on-any-espressif-esp8266-esp32/34309

GrimDeclaw, 2022-02-25:
Hi,
I just received my esp32 (wroom-32) and the web based installation just crashed on my devices.
So I decided to test the latest version from github (2.0.34) and as I already have a docker with esptool.py installed, I give it a try.
Of course, I got into the same trouble as you : how to use the esp32.bin file...
By looking at how the web installer does, I find out this file : https://espresense.com/static/esp32.json
So, I grabbed the 3 missing files (https://espresense.com/static/bootloader_esp32.bin https://espresense.com/static/partitions.bin https://espresense.com/static/boot_app0.bin) and convert all the offset to hex
4096 =>0x1000
32768 => 0x8000
57344 => 0xE000
65536 => 0x10000

After that, my esptool.py command became
esptool.py --device=/dev/ttyUSB1 write_flash -z 0x1000 /bootloader_esp32.bin 0x8000 /partitions.bin 0xE000 /boot_app0.bin 0x10000 /esp32.bin
(my ttyUSB0 is a zigbee coordinator, so beware of your config)
And that's it : esprence 2.0.34 installed ;)

Hope it helps!

blacknell, 2022-03-04:
> Hi, I just received my esp32 (wroom-32) and the web based installation just crashed on my devices. So I decided to test the latest version from github (2.0.34) and as I already have a docker with [...]

It mostly certainly did. I was pulling my hair out

DTTerastar, 2022-03-04:
https://github.com/ESPresense/ESPresense/blob/master/BUILDING.md has some info as well

primeroz, 2023-04-28:
unfortunately the `bootloader` and `partitions` file don't seem to be downloadable from those urls anymore , 404

Trikos, 2023-11-09:
I spent 3 hours in order to make this work and I finally did it.

Pre-requisites:
- PlatformIO installed
- Download and extract this project

Steps:
1. Erase the board ESP32 using esptool.py. I downloaded from [here](https://github.com/espressif/esptool) then connected the board, then launch the tool using `python esptool.py erase_flash` ( it detects automatically the port but read docs in order to avoid erasing something else)
2. Open the project ESPresense in platformIO with VS Code. It will take a while the first time you open it because it needs to download and install a lot of dependencies
3. Select the right port in platformIO
4. Click the right arrow icon on the bottom to upload the firmware
5. When you see on the VS Code terminal that it is connecting to the COM X, press both EN and BOOT buttons on the board, then release EN, wait 2-3 seconds and release BOOT. It should enter on bootloader mode and should start flashing the device.
6. When finished, go [here](https://github.com/ESPresense/ESPresense/releases) and download the latest bin version of the firmware (v3.3.1, named esp32.bin as Im writing) and rename it `firmware.bin`
7. After the build, platformIO should have created a .pio folder, look inside of it a file named firmware.bin (`.pio/build/esp32`) and replace it with the latest firmware you downloaded and renamed at step 6.
8. Repeat step 4 to upload the new firmware and update it at the latest version
9. Once finished go to [espresense.com](https://espresense.com/), connect to the board and insert the WiFi credentials, be sure to put the correct one. Even if the credentials are correct you will get a Timeout error and in order to know if it is connected you should refresh the page to free up the COM port and check on platformIO using the Serial Monitor (or whatever monitor you want) and here, after pressing EN button on the board, you can see if it will correctly connect to your wifi network.

I dont know why downloading this repo the firmware isnt updated to the latest one.
Also the web interface that you see when connecting to the board is the old one and I dont know how to change it to the last version, but It is pretty similar so.
Enjoy it

DTTerastar, 2023-11-13:
1. shouldn't be needed, and there is a button to press in pio to erase if you really need it.
5. that shouldn't be needed if your device is wired correctly.  I just hit upload and it goes on it's own
6. huh?  You've already flashed it w/ the upload button.