ESPresense/ESPresense #41: UUID beacon

April-ha, 2021-09-16:
Hello
with my Ebeacon in version 1.1.2 the UUID was upside down but visible.
the RSSI at 1m was correct and matched the beacon.

In V1.5.0 the UUIDs are no longer visible and the RSSI at 1m is blocked on -59.

https://ibb.co/M2qng8p

DTTerastar, 2021-09-16:
Yeah, I need to refactor all that power calc code.

DTTerastar, 2021-09-17:
Ok, I think it should all work now.  What rss@1m do you see now?

April-ha, 2021-09-17:
V1.5.2
same
maison = {"id":"dc3056789abc","name":"MH","rssi@1m":-59,"rssi":-75,"mac":"dc3056789abc","raw":2.87,"distance":2.26,"speed":0}

DTTerastar, 2021-09-17:
That's not the right entry.  Your beacon must be sending some TLM
broadcasts interleaved with iBeacon broadcasts.  Depending on the transmit
interval it might take a few seconds to get BOTH kinds of packets.

The iBeacon broadcast has the ***@***.*** in it, NOT the tlm broadcast.  You
should see a separate entry for the iBeacon once it finds it and that
should have a good ***@***.***

April-ha, 2021-09-17:
I did a test with "beacon simulator" on android and it works fine ("id": "iBeacon:770cddcb2d3e145784de5874ce581250-1-2").
I don't understand because my beacons worked fine in v1.1.2 and don't work after this version ("id": "dc3056789abc").
v1.1.2 works fine for me, I'll stay with this version.

DTTerastar, 2021-09-18:
Can you try master and see if it fixes it?

DTTerastar, 2021-09-19:
I ordered a Dialog based iBeacon from ali and it came today!  And it does exhibit your exact issue, and master does indeed fix it.  Releasing it now as 1.5.4.

closed 2021-09-19 by DTTerastar

April-ha, 2021-09-19:
Congratulations DTTerastar, everything works fine with version v1.5.4.
Thank you for your work

ozett, 2022-03-05:
> I ordered a Dialog based iBeacon from ali 

i am still searching for easy identify-able bluetoot devices, 
@DTTerastar : may you have a link which one you ordered?

[there are some options](https://de.aliexpress.com/wholesale?trafficChannel=main&d=y&CatId=0&SearchText=ibeacon&ltype=wholesale&SortType=total_tranpro_desc&groupsort=1&page=1), if i search for "ibeacon",
but i have no clue which to choose.. appreciate a hint .. thx.

commit e1efc112 (2021-09-18): Blind stab in the dark to fix #41
commit 0ff4650b (2021-09-19): Blind stab in the dark to fix #41