ESPresense/ESPresense #42: Same ID shown

pms0811, 2021-09-17:
I have 2 apartment entrance tag.
They all show the same ID and even all use the same name. mac address is different
People in the same apartment use the same type of tag.
So there are many of tags of the same kind.

Version:     v1.4.4
IP address:  192.168.0.14
DNS address: 168.126.63.1
Hostname:    espresense-home
Room:        home
Telemetry:   enabled
Rooms:       enabled
Devices:     enabled

{"id":"md:035d:8","name":"APT-Tag","rssi@1m":-59,"rssi":-80,"mac":"e10bb14271xx","raw":3.98,"distance":2.82:"speed":0}
{"id":"md:035d:8","name":"APT-Tag","rssi@1m":-59,"rssi":-94,"mac":"c57fc97023xx","raw":10,"distance":9.21:"speed":0}

corgan2222, 2021-09-17:
I have the same problem. There are a lot of Devices with the same ID but different MACs.
As Example, my iWatch shows the same ID as the iPhone.

> 1 Del   | MAC: 67a14bd6cf5e, ID: apple:1005:9-12
> 1 Left  | MAC: 6670b9c8f36c, ID: apple:1005:9-12                                   
> 0 New | MAC: 61913600d24f, ID: apple:1005:9-12
> 1 Close | MAC: 6020c48cf4f3, ID: apple:1005:9-12
> 0 New   | MAC: 7ee958b44127, ID: apple:1005:9-12                                   
>

DTTerastar, 2021-09-17:
> I have 2 apartment entrance tag.
> They all show the same ID and even all use the same name. mac address is different
> People in the same apartment use the same type of tag.
> So there are many of tags [...]

Is the mac randomized?  ie. If you look now are they the same mac?  Who makes those tags?  It's not a know beacon format.  The manufacture is listed as "KYS"

DTTerastar, 2021-09-17:
> I have the same problem. There are a lot of Devices with the same ID but different MACs.
> As Example, my iWatch shows the same ID as the iPhone.

See #39

DTTerastar, 2021-09-21:
I think we may have fixed this will some of the recent changes and since I haven't gotten any more info to diagnose it I'm going to close this.  Feel free to re-open as long as details about the tag type can be included.  Thanks!

closed 2021-09-21 by DTTerastar