ESPresense/ESPresense #47: OTA Update

corgan2222, 2021-09-20:
Your last update with HA Autodiscory and distance change via MQTT works great! But some of my devices are placed in not so easy to access places. 
So it would be great to have the possibility to update via OTA. Would this be possible?

DTTerastar, 2021-09-20:
I support both OTA via an auto download from github releases as well as connecting via arduino ota update.  If you use the ".bin" from github (except the -noupdate flavor) it'll automatically download and keep itself up to date.  Sometimes one or two of my devices takes a bit longer to update, not sure why.  I suspect it fails until it WDT reboots and then it works.  ESP32 isn't the most robust of environments, the WDT papers over any bugs in the platform...

corgan2222, 2021-09-21:
Ahh actually I had no idea how the OTA update works.  
But I found the documentation, so I could update the nodes. But I can't change the settings after the initial setup.
The "erase Flash" Option in PlatformIO does not work.

Maybe it would be nice, if you could write a little info about how to update and reset the ESPs.

And as I wrote in my PR, I could not found any version Information. I don't want to change too much in your code to implement it. But it would be cool to have a version number, so that Home Assistant can notify the User if a new Version is released, like for Zigbee or Hue Devices.

DTTerastar, 2021-09-21:
If you install it manually it has no version and doesn't autoupdate.  Install it from here: https://espresense.com/install and it'll auto update and will show the version

corgan2222, 2021-09-21:
Ok, understand. :)

closed 2021-09-21 by DTTerastar

spelech, 2021-10-12:
Not trying to dig up a closed issue, but could you add some instructions on using ArduinoOTA if you have some?  The Arduino IDE appears to only support OTA if you compile the project in that IDE, I couldnt find a way to use a binary compiled using platformio