ESPresense/ESPresense #57: Home-Assistant configuration to show which devices are in which rooms

phrozen77, 2021-09-23:
First off, let me apologize for my lack of knowledge regarding MQTT and Home-Assistant configuration in general...

So, there's example configuration to show which device is in which room (well, it provides the entity configuration for those).

Now, how would i go about listing the rooms in Home-Assistant and displaying the devices in each room?

Is that currently possible, and if so, how?

RichieFrame, 2021-09-24:
This would be done inside HA, probably using a template sensor (per room) that reads all the MQTT room sensors, I will see if I can make one when I have time in the next few days

corgan2222, 2021-09-24:
That's easy with this card: https://github.com/thomasloven/lovelace-auto-entities you have to install.

With this card you can filter for entity names and states:
Place a new card with this content:

```yaml
card:
  title: Devices in Office
  type: entities
filter:
  include:
    - state: office
      options:
        secondary_info: last-changed
    - name: '*espresense*'
      options:
        secondary_info: last-changed
show_empty: false
type: custom:auto-entities
```

You have to change the include filter "office" to your room name.
This should look like this:

[image]

closed 2021-10-12 by DTTerastar