ESPresense/ESPresense #58: Allow explicit inclusion/exclusion of device ID prefixes

mwhdc, 2021-09-23:
Thanks for the awesome work on this project. I just upgraded my deployment from a couple of random dev boards to a batch of Atom Lites.

Please consider adding the ability to specify the device ID prefixes that should not (or should) be reported to the MQTT server. This feature could be exposed to the user on the config page as an initially empty text field and radio buttons for exclude (default) and include. Prefix entries would be space- or comma-delimited.

I have two sets of Sonos speakers that are constantly spamming the MQTT stream with BLE pings that have no practical use and can't be disabled. I would love to be able to exclude "sonos" (i.e., sonos:*) devices from being reported. Others may want to allow only Apple devices, filter out Covid contact tracing pings, etc.

ivanvach, 2021-09-29:
I second that. It would be a very useful feature if and when implemented.

DTTerastar, 2021-10-05:
ok, so if I do this, should it be regexs or globs?

ivanvach, 2021-10-06:
Regex would be more universal and powerful but for the task at hand I think that globs will do just fine plus easier to use for non programmers.

ee02217, 2021-12-12:
This would be great. As this makes my headset change profile momentarily and thus generating a click noise :(

corgan2222, 2021-12-18:
Created a PR to add this Feature based on basic ID Filtering before sensing the MQTT Message.
https://github.com/ESPresense/ESPresense/pull/152

commit 4780e362 (2022-02-13): Add include exclude query filtering (#255)

* Add Feature: Blacklist/Whitelist
* MQTT room=* all rooms will set, added restart
* Fixes: #58 #229 #152
closed 2022-02-13 by DTTerastar