ESPresense/ESPresense #59: [Enhancement] - turn off onboard leds

ee02217, 2021-09-24:
Hi,

having these around the house is awesome! Thank you.
In my case, I use wemos ESP32, and would be great if we could disable the onboard leds to make it more discreet. 

Thank you for the great job!

markmghali, 2021-09-24:
excellent idea! That would be great enhancement!

corgan2222, 2021-09-24:
@ee02217 Do you mean this one: https://www.bastelgarage.ch/esp32minikit-wemos ?

ee02217, 2021-09-24:
That is the one :)

markmghali, 2021-09-24:
I am using these not sure if it makes a difference? There is a red and blue LED again super bright!

https://www.amazon.com/gp/product/B08D5ZD528/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

corgan2222, 2021-09-24:
As i only have the AZ-Delivery Version without the build in LED, I can't test it, but there are some lines in the code that defines the build in LED.

If you can program the ESP from code, you can try to set these line to zero
https://github.com/ESPresense/ESPresense/blob/dc4e5e37810f90b43ac1311e24e82c7a75b38620/lib/BleFingerprint/GUI.h#L38

change to
`#define LED_BUILTIN_ON 0`

DTTerastar, 2021-09-25:
All led status?  No updating status, no connecting to wifi, etc.  Or just turn off the led status of receiving btle advertisments?

markmghali, 2021-09-25:
For me I guess everything. Give a toggle to turn them all off or on? Etc

jmcclure0921, 2021-09-25:
I am not sure of the difficulty involved but the optimal light situation for me would be options for:

All Off
Flash when not connected to Wi-Fi, otherwise off
Flash when not connected to Wi-Fi and have btle adverts.

DTTerastar, 2021-09-25:
I made a huzzah32 flavor, just flashing that on a dev kit will kill all led output.  The only difference between the 2 are the led pin.

closed 2021-09-25 by DTTerastar

ee02217, 2021-09-25:
This means I can deploy adafruit-huzzah32.bin directly on my esp32?

DTTerastar, 2021-09-25:
Yep.