ESPresense/ESPresense #60: Integrate Thermal Camera

dvbit, 2021-09-24:
**Is your feature request related to a problem? Please describe.**
Integrate with Thermal Camera AMG8833

**Describe the solution you'd like**
I am currently using a ESP32 with espresense on the side of a D1Mini with Thermalcam from ThrRealWaldo
Would be wonderful to combine the 2

**Describe alternatives you've considered**
none

**Additional context**
https://github.com/TheRealWaldo/esp8266-amg8833/blob/main/src/main.cpp

DTTerastar, 2021-09-25:
Is this a good price for one of them?

https://www.digikey.com/en/products/detail/101020557/1597-1650-ND/9369930?itemSeq=377722304

JasonCCChan, 2021-09-27:
> Is this a good price for one of them?
> 
> https://www.digikey.com/en/products/detail/101020557/1597-1650-ND/9369930?itemSeq=377722304

I think you could get cheaper ones from aliexpress. I have personally tried a similar module in room-assistant. It works.

https://www.aliexpress.com/item/32963073753.html?spm=a2g0o.productlist.0.0.6ea26523S25PAT&algo_pvid=50fc58be-7009-40ca-94eb-49499f06ebd3&algo_exp_id=50fc58be-7009-40ca-94eb-49499f06ebd3-2&pdp_ext_f=%7B%22sku_id%22%3A%2266538070235%22%7D

corgan2222, 2021-09-27:
> I think you could get cheaper ones from aliexpress. I have personally tried a similar module in room-assistant. It works. [...]

Thanks for the link. I wanted to test these thermal sensors for a long time, but the prices was much higher. Ordered some of them now.

DTTerastar, 2021-09-27:
Ordered one as well

dvbit, 2021-09-27:
I used a discrete one with a d1
I think all will work

dvbit, 2021-09-27:
[image]

A combo of espresense plus thermalcam in homeassistant

Rilevato means present

DTTerastar, 2021-09-28:
How do you "interpret" the thermal camera results?

dmccabe79, 2021-10-07:
> How do you "interpret" the thermal camera results?

You look at the individual grids, which show their respective temperatures. The image just looks cool. The individual temperatures in the grid can show if there is presence in a room, which won't disappear with a lack of movement, like with a PIR.

dmccabe79, 2021-10-07:
> So if any grids are >90f you have a person? What about a sunny room?

You have to feel the room out. What's normal? What's not? Then find how it looks with somebody sitting, daylight, evening, etc

DTTerastar, 2021-10-09:
I found the algo: https://github.com/TheRealWaldo/AMG8833-ESPHOME/blob/main/amg8833_person_detected.h

closed 2022-04-10 by stale[bot]