ESPresense/ESPresense #61: Auto firmware update fails on M5StickC-Plus

StarFleetCPTN, 2021-09-24:
**Describe the bug**
Auto firmware update fails on M5StickC-Plus and causes BLE scanning to stop. Telemetry data continues to be published in MQTT after the error.

**Expected behavior**
Auto firmware update to complete and BLE scanning enabled.

**Logs**
Firmware URL:
https://github.com/ESPresense/ESPresense/releases/latest/download/m5stickc-plus.bin

Serial Log:
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():207]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -32512
[E][HTTPUpdate.cpp:231] handleUpdate(): HTTP error: connection refused

[E][main.h:205] firmwareUpdate(): Http Update Failed (Error=-1): HTTP error: connection refused

Telemetry Data:
{
  "ip": "x.x.x.x",
  "uptime": 348,
  "uptime": 363,
  "firm": "m5stickc-plus",
  "rssi": -69,
  "rssi": -68,
  "seen": 16,
  "adverts": 6120,
  "freeHeap": 63640,
  "adverts": 6355,
  "freeHeap": 63888,
  "minFreeHeap": 44604,
  "maxAllocHeap": 36056,
  "resetReason": "RtcWdtRtc"
}

**Hardware Details**
M5StickC-Plus

DTTerastar, 2021-09-25:
Yup, I missed that.  Fixed in master, will release in a bit.

closed 2021-09-25 by DTTerastar

DTTerastar, 2021-09-25:
1.6.2 is out

commit 45910ac2 (2021-09-25): Fixes #61