ESPresense/ESPresense #62: [Enhancement]

ee02217, 2021-09-25:
It would be great to have in the configuration portal a list of the seen devices as well as the ones being reported at the moment. 

This would allow to not have to use mqtt to check all devices, but see it in a convenient place. 

Tasmota has a similar feature:

[image]

ee02217, 2021-09-29:
this would allow for easier addition of new devices, at least if we had the id and the time of “first seen”

DTTerastar, 2021-10-05:
Yeah, I'm liking this idea.  It might be better than my other idea to "Join" when a device comes near.  We could instead just have a button to add to HASS Discovery.

DTTerastar, 2021-10-05:
Some devices have special abilities like BlueCharm has buttons, and motion, so that would also make it easier to specify that when adding.

closed 2022-04-10 by stale[bot]