ESPresense/ESPresense #63: Samsung devices use randomized MAC addresses

anaisbetts, 2021-09-25:
**Describe the bug**
Most newer Samsung devices (Galaxy Watch 4, >= S20) use MAC randomization. On the phone this is possible to disable but the watch it is not. Because of https://github.com/ESPresense/ESPresense/commit/d1dea7df1410fc15cc33440bae1fa342d5f555b0, Samsung now has a special case that explicitly uses the full MAC address.

**Expected behavior**
Falling back to the `md:` behavior seems to work here

**Logs**

This is what happens when you toggle phone Bluetooth, causing the watch to send extra BLE beacon packets for Where's My Watch - note that I only have a Samsung watch and phone, but there are many "unique" devices here

```
0 New   | MAC: 423f56761895, ID: samsung:423f56761895                              
0 New   | MAC: 69f5dbb6b778, ID: 69f5dbb6b778                                      
0 New   | MAC: 4dd23e16a05b, ID: samsung:4dd23e16a05b                              
0 New   | MAC: 5bafc8f35db6, ID: samsung:5bafc8f35db6                              
0 New   | MAC: 5eacd1360037, ID: 5eacd1360037                                      
0 New   | MAC: 632c7b56e42a, ID: samsung:632c7b56e42a                              
0 New   | MAC: 7ac064bc93ab, ID: samsung:7ac064bc93ab                              
0 New   | MAC: 4ded93c18123, ID: samsung:4ded93c18123                              
0 New   | MAC: 453f40f1a6ff, ID: samsung:453f40f1a6ff 
```

**Hardware Details**
Galaxy Watch 4

anaisbetts, 2021-09-25:
[image]
[image]

DTTerastar, 2021-09-25:
Yeah, samsung is a big place.  There are tvs which aren't random, so this code is good for that.  The SmartTag's are helped by this code as well.  But yeah a watch that is random that's not a good.  Is there any characteristics that make the watch identify-able?

DTTerastar, 2021-09-25:
Do you see the name in mqtt?

anaisbetts, 2021-09-25:
@DTTerastar I have active scanning turned off because turning it on destroys the battery on the watch (i.e. < 6hrs use vs ~36hrs normally). I see `samsung:[MAC]` show up in MQTT

DTTerastar, 2021-09-25:
That scan is for a "Bonded" device so it's not what we see on our side unfortunately.  I'll verify I can get what I need if you run the verbose firmware.

anaisbetts, 2021-09-25:
Here's the output with verbose firmware, doesn't seem much different:

```
0 New   | MAC: 4b50eb31169b, ID: samsung:4b50eb31169b                              
0 New   | MAC: 4e4e4e6fec47, ID: sid:0xfe9f                                        
0 New   | MAC: 03318e8c4d12, ID: sid:0xfd5a                                        
0 New   | MAC: fb3fae2ace2f, ID: md:0059:8                                         
1 Close | MAC: 03318e8c4d12, ID: sid:0xfd5a                                        
1 Left  | MAC: 03318e8c4d12, ID: sid:0xfd5a                                        
0 New   | MAC: 64a1ea9ec321, ID: exp:20                                            
0 New   | MAC: 53aed97bef30, ID: samsung:53aed97bef30                              
0 New   | MAC: 15627ea7947f, ID: apple:1307:11                                     
0 New   | MAC: 1d2637302b2c, ID: exp:20                                            
0 New   | MAC: 5e7c25e7962e, ID: samsung:5e7c25e7962e                              
0 New   | MAC: 4e8d77b19679, ID: 4e8d77b19679                                      
0 New   | MAC: 72ff4f45b423, ID: apple:1005:9-7                                    
0 New   | MAC: 007c2deb41d5, ID: samsung:007c2deb41d5                              
0 New   | MAC: 54d27264b18d, ID: apple:0631:21                                     
0 New   | MAC: 7cdc19216971, ID: apple:0f11:25                                     
0 New   | MAC: 65eb3ed281fd, ID: exp:20                                            
0 New   | MAC: 13853ead1ec1, ID: microsoft:70496e52b64f89df1aa3f0796b914963bf7534f8941e9d
0 New   | MAC: 09247c71e5a0, ID: microsoft:ac13c38f51d2df53ed9e6bde21c36e936fc6b70e5ecf7a
```

DTTerastar, 2021-09-25:
I just added the verbosity needed, are you able to flash master, or do I need to release it?

DTTerastar, 2021-09-25:
I released v.1.6.3 with this, so after autoupdating it should be in the serial output.

DTTerastar, 2021-10-01:
Any chance to look at the data?

closed 2021-10-05 by DTTerastar

bachoo786, 2022-02-08:
hey @anaisbetts @DTTerastar  I am having the same issue with my Samsung Galaxy Active Watch 4 and Samsung S20 phone. I have installed Espresense however the MAC Id for my Samsung devices keep changing.

How can I get the correct MAC ID? Can someone please help?

Thanks.

InToSSH, 2022-02-13:
I also have Watch 4. I was trying to install some iBeacon simulator app, but from what I have found the Wear OS doesn't expose an API for this. Is there any debugging I can do to see if it would be possible to get any data which would help to identify it? I noticed that when I go to BT settings, making the device discoverable I can see the name of the watch in ESPresense.

haozgit, 2022-02-24:
Complete Local Name attribute from Galaxy BLE beacon is fixed. How can leverage this as detection instead of using MAC in espresense ? Any idea ?

InToSSH, 2022-02-24:
@haozgit I found this too, but I think the Complete local name is only visible when the watch is in pairing mode or already paired to a device.

DTTerastar, 2022-03-11:
Connect w/ nRfConnect and if you can find any open services that return something trackable I can add it.  But otherwise i'm sol.

InToSSH, 2023-02-16:
> Any updates on this? Have a Samsung A32 and can't track it. It changes the MAC address like 20 times per hour and the ID is the same like MAC.

This was mainly an issue with Samsung smart watches. If you have a phone, easiest way is to install an iBeacon app which transmits the UUID and you can track that with ESPresense. This is what I am doing right now with my phone, I am using the official Home Assistant app as a beacon transmitter.

ChristianKnorr, 2023-11-01:
My solution:
- Install home assistant APP to Galaxy Watch 5 Pro
- give Access (settings/senors) to Bluetooth / BLE-Transmitter
- in espresense you get an iBeacon
- install this via https://espresense.com/home_assistant

For me it works for one ESP32, more i can't test, i ordered

xmow49, 2023-11-29:
I installed the HA app on my Galaxy Watch 4, but I don't see it...
What does the id look like?
@ChristianKnorr "give Access (settings/senors) to Bluetooth / BLE-Transmitter" this is in the HA app on smartphone? I already do that. On the watch are there any settings to change ?
Thanks

ChristianKnorr, 2023-11-29:
> I installed the HA app on my Galaxy Watch 4, but I don't see it... What does the id look like? @ChristianKnorr "give Access (settings/senors) to Bluetooth / BLE-Transmitter" this is in the HA app on smartphone? I already do that. On the watch are there any settings to change ? Thanks

In lovelace I have a sensor: sensor.christians_galaxy_watch_5_pro

InToSSH, 2023-11-30:
> I installed the HA app on my Galaxy Watch 4, but I don't see it... What does the id look like? @ChristianKnorr "give Access (settings/senors) to Bluetooth / BLE-Transmitter" this is in the HA app on smartphone? I already do that. On the watch are there any settings to change ? Thanks

In the watch HA app, go to Settings -> Manage sensors -> Bluetooth sensors -> Turn on BLE transmitter

sdetweil, 2024-01-17:
but it reports the changing mac, with the fixed UUID.. how do we define the UUID as the linkage?
```
New    | 781fe5253993 | iBeacon:b09c1e2e-8b00-45c4-8369-49cbb6479a11-100-40004
```
(from the espresense esp32 console)..
but it isn't broadcast to mqtt.. 

the UUID matches what is shown in the HA sensors, bluetooth config.  but the mac changes on every report.