ESPresense/ESPresense #64: Beacons not displaying, Base stations not registering.

hathalud, 2021-09-26:
**Describe the bug**
A clear and concise description of what the bug is.
Devices not showing in Home Assistant, neither base stations nor beacons. I followed the video and added my 4 base stations to the network and can see the traffic in MQTTExplorer. I also added the MQTT_Room sensors of the 3 devices I currently want to track to my sensors.yaml file. But in Home Assistant only 2 of my base stations registered as devices and none of my beacons are showing up at all. I have several other devices using MQTT and Tasmota, so I know it's not that. 

**Expected behavior**
A clear and concise description of what you expected to happen.
Tracked beacons and all base stations to be listed in Home Assistant

**Logs**
If possible, copy and paste any logs from the serial monitor from the time that you see the bug.
There a few errors in the HA logs

Error Number 1
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:116
Integration: MQTT (documentation, issues)
First occurred: 3:53:07 AM (6 occurrences)
Last logged: 11:41:25 AM

Unable to parse JSON max_distance: '{"~":"espresense/rooms/Kitchen","name":"ESPresense Kitchen Max Distance","unique_id":"94:B9:7E:E6:56:80_max_distance","availability_topic":"~/status","stat_t":"~/max_distance","cmd_t":"~/max_distance/set","step":"0.01","dev":{"ids":["94:B9:7E:E6:56:80"],"cns":[ESPresense Kitchen'
Unable to parse JSON max_distance: '{"~":"espresense/rooms/Livingroom","name":"ESPresense Livingroom Max Distance","unique_id":"7C:9E:BD:61:F9:2C_max_distance","availability_topic":"~/status","stat_t":"~/max_distance","cmd_t":"~/max_distance/set","step":"0.01","dev":{"ids":["7C:9E:BD:61:F9:2C"],"cns":['
Unable to parse JSON connectivity: '{"~":"espresense/rooms/Livingroom","name":"ESPresense Livingroom","unique_id":"7C:9E:BD:61:F9:2C_connectivity","json_attr_t":"~/telemetry","stat_t":"~/status","frc_upd":true,"dev_cla":"connectivity","pl_on":"online","pl_off":"offline","dev":{"ids":["7C:9E:BD:61:F9:2C"],"cns":[ESPresense Livingroom'
Unable to parse JSON max_distance: '{"~":"espresense/rooms/Livingroom","name":"ESPresense Livingroom Max Distance","unique_id":"7C:9E:BD:61:F9:2C_max_distance","availability_topic":"~/status","stat_t":"~/max_distance","cmd_t":"~/max_distance/set","step":"0.01","dev":{"ids":["7C:9E:BD:61:F9:2C"],"cns":[ESPresense Livingroom'
Unable to parse JSON 1F28A4_RL_3: '{"name":"Plug2-3","stat_t":"tele/Plug2/STATE","avty_t":"tele/Plug2/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/Plug2/POWER3","pl_off":"OFF1(&homeassistant/light/1F28A4_LI_6/config1)'homeassistant/switch/1F28A4_RL_6/config1(&homeassis'

Error Number 2
Logger: homeassistant.components.mqtt
Source: components/mqtt/__init__.py:895
Integration: MQTT (documentation, issues)
First occurred: 11:12:16 AM (2 occurrences)
Last logged: 11:51:14 AM

Can't decode payload b'{"id":"iBeacon:64d5478b-d3ec-436b-83c9-967ed3dd87df-100-1","rssi@1m":-59,"rssi":-70,"mac":"40d2f2139790","raw":2.06,"distance":1.4,"spee\x82*\x00\x04\x00' on espresense/devices/iBeacon:64d5478b-d3ec-436b-83c9-967ed3dd87df-100-1/Haths Bedroom with encoding utf-8 (for <Job HassJobType.Callback <function MQTTRoomSensor.async_added_to_hass.<locals>.message_received at 0x7f7241d0bb80>>)
Can't decode payload b'{"~":"espresense/rooms/Haths Bedroom ","name":"ESPresense Haths Bedroom ","unique_id":"7C:9E:BD:62:18:E0_connectivity","json_attr_t":"~/telemetry","stat_t":"~/status","frc_upd":true,"dev_cla":"connectivity","pl_on":"online","pl_off":"offline","dev":{"ids":["7C:9E:BD:62:18:E0"],"cns":[\x86\xe0\x07\xc0\xbc\xd8\xff?' on homeassistant/binary_sensor/espresense_Haths Bedroom /connectivity/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f7241f4e790>>)

Error Number 3
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:101
Integration: MQTT (documentation, issues)
First occurred: 12:07:21 AM (103 occurrences)
Last logged: 12:10:35 PM

Received message on illegal discovery topic 'homeassistant/number/espresense_Haths Bedroom /max_distance/config'
Received message on illegal discovery topic 'homeassistant/number/espresense_Kyras Bed/max_distance/config'
Received message on illegal discovery topic 'homeassistant/binary_sensor/espresense_Kyras Bedroom/connectivity/config'
Received message on illegal discovery topic 'homeassistant/number/espresense_Kyras Bedroom/max_distance/config'
Received message on illegal discovery topic 'homeassistant/binary_sensor/espresense_Haths Bedroom /connectivity/config'

**Hardware Details**
If the bug is related to tracking a specific piece of hardware (e.g. iBeacon, phone, etc), please enter any relevant details about the hardware (such as firmware version, MAC address, etc).
My setup for the beacons:
- platform: mqtt_room
  device_id: "38854575b6e4"
  name: 'Esp Haths Nut'
  state_topic: 'espresense/devices/38854575b6e4'
  timeout: 6
  away_timeout: 20

- platform: mqtt_room
  device_id: "iBeacon:64d5478b-d3ec-436b-83c9-967ed3dd87df-100-1"
  name: 'Esp Haths S21'
  state_topic: 'espresense/devices/iBeacon:64d5478b-d3ec-436b-83c9-967ed3dd87df-100-1'
  timeout: 6
  away_timeout: 20

- platform: mqtt_room
  device_id: "iBeacon:64f8b12b-6c89-4969-a829-d4d1f9e3b716-100-1"
  name: 'Esp Kyras S21'
  state_topic: 'espresense/devices/iBeacon:64f8b12b-6c89-4969-a829-d4d1f9e3b716-100-1'
  timeout: 6
  away_timeout: 20

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]
[image]
[image]

hathalud, 2021-09-26:
[image]

DTTerastar, 2021-09-27:
I have seen this happen.  After a reboot it usually doesn't reoccur.  I haven't figured out exactly what causes it to happen.  I think I need to try do the cns a different way, it seems like the buffer get's deleted before it's serialized sometimes.

hathalud, 2021-09-27:
I have restarted Home Assistant and all of the ESPresense devices. The entities for the beacons I want to track are in HA now, but the same ESPresense devices are missing. 

Not being a programmer in a loooong time I can't give too much feedback or suggestions, but is it possible to delay acting until the buffer you are referring it is serialized?

corgan2222, 2021-09-27:
> 
> 
> [image]

Is there a whitespace at the end of "Haths Bedroom" in the first and last line in the screenshot? 
I had some problems with Name which includes whites paces, so I try to avoid this on the lower level like hostnames, mqtt, aso.
In Home Assistant you can rename everything later.

hathalud, 2021-09-27:
@corgan2222 Oh, good catch. That explains why the binary sensor kept reporting that particular base station as offline even though it wasn't. I think that one I setup from my phone, which explains the whitespace. Thanks.

The other issues are still persisting. I at least have the beacon entities showing up in HA but not the other base stations.

corgan2222, 2021-09-27:
Nice! :)
I would suggest to start from scratch with completely new names without white space. As I tested the new sensor integrations, I had some invalid leftover stuff in the MQTT broker, which totally confused the HA handling of it.

hathalud, 2021-09-28:
I used MQTTExplorer and deleted the wrongly named topics. I might try
rebuilding in a day or two if I don't hear anything else on the issue soon.

DTTerastar, 2021-09-28:
Can you see if 1.6.4 fixes the issue for you?

hathalud, 2021-09-28:
Assuming that the web browser install is using that version, I just updated
the two ESPs that were not showing in HA. Doesn't seem to be making any
difference. I tried deleting their topics from the broker and that didn't
have any impact either. Doing a clean install with one of them and a
different room name. Still no success.

A few suggestions though:

   1. Display the current version in the web interface (WGUI)
   2. Offer a means of manually updating or initiating an OTA update of the
   firmware from the WGUI
   3. Is it possible to load the web terminal from the WGUI?

DTTerastar, 2021-09-29:
Can you look in home assistant for the discovery errors?  That would help.

hathalud, 2021-09-29:
Of course.

Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:101
Integration: MQTT (documentation, issues)
First occurred: 2:07:39 PM (26 occurrences)
Last logged: 9:39:19 PM

Received message on illegal discovery topic
'homeassistant/number/espresense_Kyras Bedroom/max_distance/config'
Received message on illegal discovery topic
'homeassistant/number/espresense_Haths Bedroom/max_distance/config'
Received message on illegal discovery topic
'homeassistant/number/espresense_Kyras Bedroom V2/max_distance/config'
Received message on illegal discovery topic
'homeassistant/binary_sensor/espresense_Kyras Bedroom
V2/connectivity/config'
Received message on illegal discovery topic
'homeassistant/binary_sensor/espresense_Haths Bedroom/connectivity/config'

And here's the message content:
{"~":"espresense/rooms/Kyras Bedroom V2","name":"ESPresense Kyras Bedroom
V2 Max
Distance","unique_id":"7C:9E:BD:62:1A:A8_max_distance","availability_topic":"~/status","stat_t":"~/max_distance","cmd_t":"~/max_distance/set","step":"0.01","dev":{"ids":["7C:9E:BD:62:1A:A8"],"cns":[["MAC","7C:9E:BD:62:1A:A8"]],"name":"ESPresense
Kyras Bedroom V2","sa":"Kyras Bedroom V2","mdl":"ESP32-D0WDQ6"}}

DTTerastar, 2021-09-30:
OK I added the slugify code so you can now use spaces successfully in room names which should fix the above.

hathalud, 2021-09-30:
Yup. That fixed the issue. I didn't even have to do anything, as they
updated themselves. Thank you.

commit 9d2bf69e (2021-09-30): Fixes #64
closed 2021-10-01 by DTTerastar