ESPresense/ESPresense #65: Enhancement requests for M5Stack devices (M5StickC, M5StickC-Plus, M5Atom-Matrix)

mcsquared88, 2021-09-27:
All of these M5Stack devices support the I2C protocol over a Grove port, and M5Stack makes a Grove PIR sensor. It would be great if ESPresence supported a PIR sensor via this interface.

Another suggestion that isn't specific to the M5Stack devices: A "silent" option in the configuration webUI to disable the screen and LED flashing. This would make using these devices on a nightstand a lot less intrusive.

DTTerastar, 2021-09-27:
If you don't want the screen or led, just flash the esp32 variant and it'll all turn off (once you power it on and off).

DTTerastar, 2021-09-27:
https://www.digikey.com/en/products/detail/m5stack-technology-co-ltd/U004/12088549 is probably not supported (I don't have one to test), but I think this will work with the existing support: https://www.digikey.com/en/products/detail/U054/2221-U054-ND/12088559?itemSeq=377364194

DTTerastar, 2021-09-27:
Actually I take that back, both should work already just fine.  Try putting 38 in pirPin.  These aren't actually i2c.  Can yoou test and close this?

mcsquared88, 2021-09-27:
Excellent suggestion about just flashing the ESP32 variant. I'll try that. I do have the Grove/I2C PIR sensor and I'll test whether that works w/ the M5StickC and ATOM Lite devices I have. I'll report back here once I've done the testing.

DTTerastar, 2021-09-28:
Hmm, there might be some init code we're missing then.

DTTerastar, 2021-10-05:
Any more info?  I looked through the m5 init code and don't see any explanation for what you reported.  Should we close this?

DTTerastar, 2021-10-11:
I'm going to close this for now.  You can open another issue if you figure out what's the difference.