ESPresense/ESPresense #68: How-to connect motion sensor to esp32?

markmghali, 2021-09-28:
This may not be the correct place but I want to wire a AM312 Motion sensor to my esp32 how would I go about doing that with this project? 

This is the sensor 
https://www.amazon.com/Weewooday-Pyroelectric-Infrared-Detector-Temperature/dp/B08PVB7VHQ/ref=sr_1_5?dchild=1&keywords=am312+motion+sensor&qid=1632842813&s=electronics&sr=1-5

DTTerastar, 2021-09-28:
This PR has a diagram: https://github.com/ESPresense/ESPresense/pull/48 
We need to get that added to the docs.  Any help would be appreciated in https://github.com/ESPresense/ESPresense.github.io

markmghali, 2021-09-28:
Yeah this diagram is for the Radar sensor? I have vague understanding of how to wire it I believe.

DTTerastar, 2021-09-28:
I think the pir is the same.  It's just a HIGH or a LOW coming into a pin.  Just make sure you only let in 3.3v to the pin or else your ESP32 will be unhappy.

DTTerastar, 2021-09-28:
@corgan2222 can you help out?

markmghali, 2021-09-28:
Oh not sure how to regulate the voltage to 3.3 maybe I'm in over my head?
Haha

corgan2222, 2021-09-28:
Your mini AM312 runs at 3V. I only use the HC-SR501 which runs at 5V.

you can look here how to conect;
https://randomnerdtutorials.com/esp32-pir-motion-sensor-interrupts-timers/

markmghali, 2021-09-29:
You guys have been very helpful! I will just get the 5v one so I can wire it straight to the esp32!

Thank you

closed 2021-09-29 by markmghali

DTTerastar, 2021-10-01:
I was able to easily hook a AM312 up to my m5atom.  There is a 3.3v pin I used for supply, and ground, and then I hooked the "OUT" to gpio 33.  It seems to work great atm.

markmghali, 2021-10-01:
Awesome thank you for the follow up. I think I see a 3v3 pin my esp32 I could try and use