ESPresense/ESPresense #70: Add Lux Sensor

markmghali, 2021-09-29:
is it possible to add a lux sensor like you did with the Motion sensor? Then this would be the ultimate detection machine!

corgan2222, 2021-09-29:
Which sensor do you want to use?
I have this one https://www.amazon.de/gp/product/B07DDN5TH2/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 which I could test with.

If @DTTerastar gives his go, I could implement this.

markmghali, 2021-09-29:
I don't have a specific one in mind. Whatever works best with it!

I found it on US Amazon as well

Songhe LM393 Light Detection Optical Sensitive Resistance Sensor Module
Photosensitive Sensor for Arduino 3 pin (10PCS)
https://www.amazon.com/dp/B07XFZ99XL/ref=cm_sw_r_apan_glt_fabc_KJGN7A25HZYN40Z1BRJE

corgan2222, 2021-09-29:
I think that's the same sensor as I have here. 
And because the sensor is on the PCB, It's easy to read the digital values.

corgan2222, 2021-09-29:
I just looked at the data sheet and noticed that I was wrong. 

The LDR5528 and your LM393 can be used as a kind of light switch. If a defined (with the knob) threshold is reached, the sensor returns a 0 or 1. Like the bulb in the fridge. :)

But i think you want to know the lux level to use this value in automations?

corgan2222, 2021-09-29:
The most common models are the TSL2561 
https://esphome.io/components/sensor/tsl2561.html

and the BH1750 
https://esphome.io/components/sensor/bh1750.html

both uses I2C.

markmghali, 2021-09-29:
Correct! Ambient light sensor is what I was looking for!

corgan2222, 2021-09-30:
I ordered some of these sensors. Will check into this next week.
Hope that @DTTerastar are fine with supporting more and more sensors. For me, it makes perfect sense. :)

markmghali, 2021-09-30:
This is really what I was going for something similar to this but with
espresense

https://esphome.io/cookbook/bruh.html

corgan2222, 2021-09-30:
HRH, I use ESP Home for some other things, but never heard of this before. 
Yes, that's exactly what I want to have in each room. But with some more options and a much cooler enclosure, which I'm currently working on.

markmghali, 2021-09-30:
Oh I don't have a 3d printer but I would love to see what you come up with
and maybe pay to get them printed at some point!

corgan2222, 2021-09-30:
It a rework of this Design

sadly it made only for the ESP8266 and did not have enough space for the ESP32. So i have to rebuild it.

[image]

With all parts, it should look like this

[image]

It should provide space for
- Temp/Humidity
- Pir
- Radar
- LED

all sections can be standalone, so it can be expanded later.

markmghali, 2021-09-30:
Oh that is super cool!

This is the design for the Bruh Multi sensor
but yes like you said not enough space for esp32

https://www.thingiverse.com/thing:3538811

corgan2222, 2021-09-30:
Yeah, I have tons of these style of cases downloaded from thingiverse.
But I want to have a more fancy design, as these things should stay in every room of our house. 
And that's the perfect time to learn more advanced things in Fusion 360 I wanted to learn more for a long time, but didn't had a project for. :)

DTTerastar, 2021-09-30:
I'm okay to a point,  the sensors need to at lease somewhat be related to presence... Lux is important to presence since you use it to prevent occupancy...  I do think the weather isn't really all that related, but someone did the work, so oh well let it be.

DTTerastar, 2021-09-30:
This is the sensor I want implemented: https://www.youtube.com/watch?v=c91Ve-g0J2U

DTTerastar, 2021-09-30:
I'm hoping this one will also work: https://shop.m5stack.com/collections/stick-series/products/m5stickc-tof-hatvl53l0x

corgan2222, 2021-09-30:
Ok, cool. If the lux sensors arrive next week, I will implement it. :) 

These Tof sensor looks very cool! Have to check later if these are also available as normal sensors. 
Or maybe I get one of these fancy M5 things. :) Never have looked into it, but they look interesting.

corgan2222, 2021-09-30:
Just ordered those VL53L0X :)
https://www.amazon.de/AZDelivery-VL53L0X-Flight-kompatibel-Arduino/dp/B086TSKJBT/ref=pd_sbs_2/262-7871613-8793545?pd_rd_w=UZvCw&pf_rd_p=b2a25e73-6e6a-41bb-a7ce-5f2d2342b819&pf_rd_r=62EB05S8GQ2XR1DJH81S&pd_rd_r=f50f0f86-c85d-418b-9469-8bfd0d6246ed&pd_rd_wg=JjDLI&pd_rd_i=B086TSKJBT&psc=1

markmghali, 2021-09-30:
Awesome, very curious how this will all shape up!

Thank you guys for your work on this

TarcisioMenezes, 2022-01-19:
Tried to find some documentation about, but I couldn't. How can I configure the sensors supported by ESpresence? (PIR, LUX e etc...)?
Thank you!

YBonline, 2022-07-13:
Is there any guide to how to enable this sensor? I just bought one and realized there seems be no configuration anywhere, from what I can see?

DTTerastar, 2022-07-13:
> Is there any guide to how to enable this sensor? I just bought one and realized there seems be no configuration anywhere, from what I can see?

Makes sure you're installing the version w/ sensors.  Scroll down in the dropdown and you should see it.