ESPresense/ESPresense #71: Take the intelligence to the edge of the network

ivanvach, 2021-09-29:
As the name suggests this is "Presense" detection.
Make it act like one by

1. Allow user to weed out unnecessary for the presence detection BT devices like stationary computers, BT speakers etc.
2. Analyze presence on board of ESP32 and produce MQTT record only when one of the tracked devices enters or leaves the zone determined by radius in the setup.

The way it is now  ESPresence produces a barrage of MQTT records (most unnecessary) that some computer(s) upstream have to read and analyze. ESP32 can and must do this analysis locally not clogging the network and upstream devices with information that can be processed locally.

corgan2222, 2021-09-30:
I fully agree. @DTTerastar seams to working on it allready. https://github.com/ESPresense/ESPresense/issues/53 https://github.com/ESPresense/ESPresense/pull/67

For me personally, a simple whitelist would do the job perfectly fine.

DTTerastar, 2021-09-30:
ESP32s aren't a full linux environment.  They can't do nearly as much as a pi.  We only have a few kbs of memory!  Also, C++ is a bad place to try a build a ton of functionality.  I do think I can add some filtering, but I don't think it's really needed. Mqtt passing a few extra packets is no big deal.  I attempted to have the esp listen for packets AND request a gap characteristic and it totally broke the esp... So treating them more like sensors nodes and using real smarts in python is more the direction I see this project going... But hey PRs are accepted if you have the programming chops.  I'm really a c# programmer, not a C++ guy this is totally new to me.

DTTerastar, 2021-09-30:
The other thing is the "barrage" is mostly because of noise reading rssi.  You really need multiple nodes to correlate the rssi and figure out if the rssi is real or noise.  If we could filter out all the rssi readings that are crap, we wouldn't need to report much into mqtt.

ivanvach, 2021-09-30:
Yes. C# and C++ are quite different. Congrats on making the transition.

DTTerastar, 2021-10-05:
I did find a bug that made the code to send more traffic than desired.  The latest release honors the setting to skip sending an update if the position is < whatever distance you specify.

ivanvach, 2021-10-05:
That's great Darrel !!! Thanks a lot. I will test it today and report back.

ivanvach, 2021-10-05:
Noticed all great improvements but this particular problem is not resolved yet IMHO.  I set it up very conservatively to report moves 7+ meters  ONLY and to update in 20 sec.
[log1.txt](https://github.com/ESPresense/ESPresense/files/7288595/log1.txt)
. Still it produced 2500+ MQTT messages in 15 minutes that didn't contribute to presence detection.  I did move the only BT device that needs monitoring once and ESPresence didn't register when it left the radius of 16 m. It did register the return. 
My office where I test is not different than many others. I have few computers, a printer, Google Home speaker, few tablets but as I said I was moving only one. Everything else was stationary.
Part of MQTT log with time stamps is attached if someone wants to analyze it.
[log1.txt](https://github.com/ESPresense/ESPresense/files/7288601/log1.txt)

DTTerastar, 2021-10-05:
This is made to work with https://www.home-assistant.io/integrations/mqtt_room/

The rules of using that are you send *nothing* if > maxDistance, and you keep on sending messages if < maxDistance.  I do agree that it's probably the worse way of using mqtt if you only have one base station.  But the whole point of `mqtt_room` is to have hass figure out what rooms something is in.

ivanvach, 2021-10-05:
Actually your reports (log) to the serial port are much nicer and useful than the MQTT.

1. It reports the arrival once only as "new".
2. It automatically removes the device as "del" if it didn't hear from it for certain time. 
3. When the device comes back it is reported as "new" again.
4. Best of all it produced only 100 records in my 15 minutes test vs 2,500 useless MQTT records.

1. to 3. above is sufficient to determine the presence in certain area. Please note that PIR sensor marks arrival as well. However if I "arrive" and lay on the couch for 2 hours reading a book than PIR senor will "think" that I'm gone and will turn off the lights. In this regard I wish that ESPresence is better and marks arrival as well as staying, which ESPresence does over serial interface but not over MQTT. Your point that object will move from one to another monitored area is not very strong as the object may leave the house as well and leave the system in believe that it is in the last area.

ivanvach, 2021-10-05:
Can you please do this:

1. send MQTT JSON ONLY when you send out the serial log records. 
2. Include the important info "new' or "del" in the JSON 

If someone likes the MQTT clutter than make the above optional.

DTTerastar, 2021-10-05:
No.

closed 2021-10-05 by DTTerastar