ESPresense/ESPresense #73: Integration with ESPHome

travipross, 2021-10-04:
**Is your feature request related to a problem? Please describe.**
Is there any way to have this project play more nicely with ESPHome? I find that I want two ESP32's in each room because of features on both of these projects.

**Describe the solution you'd like**
Perhaps ESPHome compatibility built into this project such that various sensors and outputs can be interacted with through the ESPHome ecosystem? Or an official plugin for ESPHome from the contributors of this project?

**Describe alternatives you've considered**
2 ESP32s in one room.

**Additional context**
None

FM-17, 2021-10-04:
I'm hoping to add ESPresence-enabled ESP32's to my home for pet/asset tracking as well as ESPHome-enabled ESP32's for temp + humidity monitoring and PIR motion detection. Using two ESP32's per room for this would be wasteful, so I too would love to see this added.

corgan2222, 2021-10-04:
> 
> 
> I'm hoping to add ESPresence-enabled ESP32's to my home for pet/asset tracking as well as ESPHome-enabled ESP32's for temp + humidity monitoring and PIR motion detection. Using two ESP32's per room for this would be wasteful, so I too would love to see this added.

That's exactly the reason I implemented temp + humidity + pir + radar sensor support into ESPresence. :)
https://github.com/ESPresense/ESPresense/pull/54
https://github.com/ESPresense/ESPresense/pull/48

coming this week:
- lux sensor

If you need other sensors, just let us know.

travipross, 2021-10-05:
> That's exactly the reason I implemented temp + humidity + pir + radar sensor support into ESPresence. :)

@corgan2222 Awesome, is there any documentation you can refer me to in order to learn more? I wasn't able to find any mention of this functionality in the ESPresense docs.

Though this is a fantastic addition to this project, ESPHome still has a considerable benefit of its huge catalog of supported sensor types, and its flexible configuration approach. So I'll certainly plan to make use of your recently added functionality, but I'll still keep this feature request open to hopefully encourage some more discussion on the topic and see if deeper integration might be possible. 

PS. The DHT11 and DHT22 are some of the most popular temperature sensors out there, but they're not widely regarded as the most accurate ([here is one random article](https://home-assistant-guide.com/2020/10/05/the-best-temperature-and-humidity-sensor-for-esphome/) outlining some comparisons). Would be great to see [BMP280](https://esphome.io/components/sensor/bmp280.html) / [BME280](https://esphome.io/components/sensor/bme280.html) support added here as a holdover ;)

FM-17, 2021-10-05:
> That's exactly the reason I implemented temp + humidity + pir + radar sensor support into ESPresence. :) #54 #48
>
> coming this week:
>
> * lux sensor
>
> If you need other sensors, just let us know.

Thanks very much for the context. @travipross has touched on most of my follow up comments already.

I really appreciate that you've implemented those sensors to hold us through in leiu of ESPHome integration. Though I'd still love to see ESPhome integrate with ESPresense someday (or vice-versa) to reduce the amount of recurring development required for sensor integration.

corgan2222, 2021-10-05:
> @corgan2222 Awesome, is there any documentation you can refer me to in order to learn more? I wasn't able to find any mention of this functionality in the ESPresense docs.
>
> Though this is a [...]

ATM there are only these both push-request where you can find some infos.
But I have planned to write some kind of documentation this week. 
I have to look, if i have some of these BMP280 / BME280 somewhere in my boxes.

I have zero knowledge about how ESPHome works and how hard is it, to implement ESPresense into it. 
Every time i tried to do something with it, I wanted to  toss it out of the window. So I personally don't have plans to spend time on this. Actually its @DTTerastar project, maybe he wants to take a look.

But I can absolutely understand your needs to have only one device in each room, what can do everything. That was the reason to implement the sensors in the first place.

DTTerastar, 2021-10-06:
I already have loads of zigbee motion, lux, and temp sensors... I do agree that FAST motion is something that Zigbee doesn't fully do well.  I got the radar sensors, need to hook them up and see if they are better than pir.

Heisenberg2980, 2021-10-30:
> BME

Another vote for the addition of BME280 to this amazing project. If it helps, at the moment I am using ESP32-mqtt-room (PlatformIO) installed in ESP32 devices around the house for room detection, and I added a BME280 to one of them, this is the part I added to report the temperature of the sensor (happy to share the entire code if needed):

```
bool sendBME280() {
	if (debug) mqttClient.publish(debugTopic, 0, 0, "sendBME280 start");
	StaticJsonDocument<256> BME280;
	BME280["temperature"] = bme.readTemperature();
	BME280["humidity"] = bme.readHumidity();
	BME280["pressure"] = bme.readPressure() / 100.0F;

	char BME280MessageBuffer[258];
	serializeJson(BME280, BME280MessageBuffer);

	if (mqttClient.publish(temperatureTopic, 0, 0, BME280MessageBuffer) == true) {
		Serial.println("BME280 info sent");
		if (debug) mqttClient.publish(debugTopic, 0, 0, "sendBME280 end");
		return true;
	} else {
		Serial.println("Error sending BME280 info");
		mqttClient.publish(errorTopic, 0, 0, "Error sending BME280 info");
		if (debug) mqttClient.publish(debugTopic, 0, 0, "sendBME280 end");
		return false;
	}
}
```

corgan2222, 2021-10-30:
This one:  https://www.amazon.de/Hojffuue-Temperatur-Feuchtigkeit-AtmosphhRischer-Barometrisch/dp/B09CYCT4CN/ref=sr_1_1_sspa?adgrpid=72630429073 ?

Heisenberg2980, 2021-10-30:
> This one: https://www.amazon.de/Hojffuue-Temperatur-Feuchtigkeit-AtmosphhRischer-Barometrisch/dp/B09CYCT4CN/ref=sr_1_1_sspa?adgrpid=72630429073 ?

No, that is the BMP280, the one I have is the BME280: https://www.amazon.de/BME280-Feuchte-Temperatur-Atmosph%C3%A4rendruck-Sensor-Modul-mit-IIC-Arduino/dp/B09CYXS3GX

I think the main difference between them is the BMP280 can only measure temperature and air pressure, while the BME280 can measure humidity in addition to temperature and air pressure.

corgan2222, 2021-10-30:
Ah, ok, I see. Didn't realize the BME/P difference. 
Here in Germany the single sensor is with €15 quite expensive compared with the BMP280.  There is one for only €5 but can't deliver here. Will look on Ali later.

Heisenberg2980, 2021-10-31:
Ok, I guess you need the sensor to be able to test it, but just in case if you want to add the code before you receive the sensor, I am happy to test it for you

DTTerastar, 2021-11-03:
I ordered a sensor as well...

mloebl, 2021-11-03:
I have looked to do custom boards with BMP280, but it's a pain as looks like either have to buy a break out board (like from Adafruit and others), or know a distributor in China willing to sell them. If you don't care about pressure, check out the SHT40 as well.

mjcumming, 2021-11-06:
I have the lux sensor working great. The DHT22 is not. I see it on the setup page, but is it working?

mjcumming, 2021-11-07:
> > > I'm hoping to add ESPresence-enabled ESP32's to my home for pet/asset tracking as well as ESPHome-enabled ESP32's for temp + humidity monitoring and PIR motion detection. Using two ESP32's per room for this would be wasteful, so I too would love to see this added.
> > 
> > 
> > That's exactly the reason I implemented temp + humidity + pir + radar sensor support into ESPresence. :) #54 #48
> > coming this week:
> > 
> > * lux sensor
> > 
> > If you need other sensors, just let us know.
> 
> With PIR, lux, DHT and this... it's literally the perfect multi sensor. Just need a decent 3d printer model and a place to run power. Gotta smash the poll times on these cool sensors, no time for batteries. :)

If you are looking for a great plugin case, I use these

https://www.polycase.com/plug-in-wall-plug-enclosures

corgan2222, 2021-11-07:
> If you are looking for a great plugin case, I use these
> 
> https://www.polycase.com/plug-in-wall-plug-enclosures

I have searched for this kind of plug/enclosure for Europe 220v, but couldn't find it. If you see the kind of plug for 220v, please let us know.

cpainchaud, 2021-11-12:
@travipross @FM-17 @corgan2222 given that BLE presence tracker exists on ESPHome: https://esphome.io/components/binary_sensor/ble_presence.html
This is all about adding support for rooms to send MQTT specific messages? or may be the ESPHome already does that? I need to give it a try

travipross, 2021-11-12:
@cpainchaud ESPresense also does MAC fingerprinting (which is necessary for devices that show up with variable MAC addresses) and adds a kalman filter (which should greatly improve accuracy). 

As far as I can tell, ESPHome doesn't do any of that, so it would work less effectively and on a lesser number of devices.

mloebl, 2021-11-12:
I have to admit, from a security standpoint, ESPHome long term may be a safer place to be? Right now there's really no password protection or ssl, right? So anyone on the network can connect and make changes to ESPresense. What "keeps me up at night" is that the auto-update ability is awesome and I've been lazy and left it enabled, but if someone compromises an account for github and pushes a nefarious build, the devices will just run with it? My other option is to leave it disabled, then having to go to each device individually to update vs a push from Home Assistant. Just some things to think about that may or may not be warranted. 

I love ESPresense, but I think in terms of management, unless the ESPresense team wants to write management abilities and deal with security, ESPhome could be a good place for it to be long term replacing or complimenting the limited ESPHome BLE abilities it currently has? Not looking to start a philosophical battle as truly appreciate the work the work the team here is doing, but just my .02.

mjcumming, 2021-11-13:
While I use ESPHome, I do not think integration is worthwhile. ESPresence is so simple to deploy and manage. There is no need to add the complexity of ESPHome. ESP32 devices are so cheap that trying to create a platform that does everything is not needed.

mloebl, 2021-11-13:
> While I use ESPHome, I do not think integration is worthwhile. ESPresence is so simple to deploy and manage. There is no need to add the complexity of ESPHome. ESP32 devices are so cheap that trying to create a platform that does everything is not needed.

I can see that but not sure I fully agree, I've got quite a few ESPHome devices around the house, and super simple to set them up and arguably takes less time as I don't need to connect via Wifi to configure them after being programmed (just create config, plug it in, upload and done). You end up with a device that's easy to manage, update, and "secure" (well relatively speaking.) 

I'd actually argue the opposite is happening with feature creep in ESPresense as it's adding features like sensors and such which is making it more like ESPHome every time.  This could be the goal of the project long term? I don't know. What worries me what will happen is there's going to be more and more "Can you support this sensor?" and ESPresense is going end up with kind of an unweildy interface losing what it was meant for and the developers are going to burn out trying to recreate what's already out there.

FWIW I am the kind of person that likes to DIY with a feeling of "I have to do this myself" and create and program for embedded boards (I've rolled quite a few custom boards in my day from PICs to Atmega board thru to the ESP32-S2). However starting to realizing things like ESPHome, Tasmota and Espurna eliminate the need for 1000s of variations on the same exact theme and eliminate projects dying because developers get burnt out trying to make everyone happy trying to add everything.

travipross, 2021-11-13:
>no need to add the complexity of ESPHome
@mjcumming 

Perhaps you're correct that it doesn't make sense to see all of ESPHome's functionality duplicated in this project. This project does one thing very well, while ESPHome has an enormous catalog of features that work very well, plus it includes a number of other features including [encryption via native API instead of MQTT](https://esphome.io/components/api.html#native-api-component), [Password protected OTA updates ](https://esphome.io/components/ota.html?highlight=password#ota-update-component), a [management dashboard interface](https://esphome.io/guides/getting_started_hassio.html#dashboard-interface), and many others.

Like @mloebl, I don't want to diminish any of the work of the devs on this project, as they've created some fantastic functionality. I just wonder if that functionality would be best ported over to be compatible with an already well-established ecosystem for ESP devices, one with an already large community of enthusiastic users and developers.

The MAC fingerprinting and kalman filtering in this project would make the [ESPHome BLE component](https://esphome.io/components/binary_sensor/ble_presence.html?highlight=presence%20ble#esp32-bluetooth-low-energy-device) a suitable replacement, allowing users to have the best of both worlds.

Heisenberg2980, 2021-11-13:
I completely agree with @travipross, combining the best of both worlds by adding the extra features of ESPresence into ESPHome would be the prefect solution (unless for some reason this is impossible). If I was one of the ESPresence developers I would try to do this rather than burn myself out by trying to add hundreds of ESPHome features/sensors into ESPresence.

mjcumming, 2021-11-13:
Good discussion - pros and cons with both approaches.  The simplicity of this project, with web-based programming of the ESP  is a great feature.  Adding a few simple sensors for lux, temp, etc is an added bonus. ESP Home is much more complex and intimidating. 

I think the real win for this project is developing the logic for better device location determination through training with a network of ESP BT sensors.

mloebl, 2021-11-13:
> Good discussion - pros and cons with both approaches. The simplicity of this project, with web-based programming of the ESP is a great feature. Adding a few simple sensors for lux, temp, etc is an [...]

Totally agree, I think as mentioned above, the best case scenario would be if we could have both the update to ESPHome and a standalone ESPresense using the same code base for the BLE Detection.

To be completely honest, I much prefer how ESPresense handles trackers at the MQTT level and sending any found devices than when I last holooked into ESPHome at it required it at the device level it looked like (so you add a whitelist of MAC addresses at each ESPHome device.)  However I really miss the central ability of management I get with ESPHome. I am being greedy I guess and want both :D

FM-17, 2021-11-13:
I completely agree with the idea that ESPresense has advantages over ESPhome, this was likely the motivation for suggesting their collaboration initially. It is a shame that the two cannot coexist on one ESP32 though, as I'd love to make use of ESPresense's accuracy without sacrificing any of the ESPHome features im currently using. 

On the other hand, simplicity is nice, why change what is already working for some? 

Sure, ESPHome is modular and therefore only as complex as you make it, but not all users want to spend an additional two hours climbing the learning curve just to gain some flexibility. Similarly, its not realistic to expect those who appreciate ESPresense's simplicity to maintain a fork of it while the main branch becomes cog in a much larger, more comprehensive system. 

ESPresense was fun to experiment with, I'm going to miss it after switching back. I just hope that when ESPHome does implement similar functionality, it's done in collaboration with the amazing ESPresense developers, as I'd hate to see such great work be overshadowed by an inferior implementation.

cpainchaud, 2021-11-14:
By creating a 'presense' library which would host the core of the logic,
both projects could benefit from this great work

sourbrambles, 2021-11-23:
Esphome is working it's way towards simplification for new users (Web and Bluetooth initial config, easy to share configuration etc.), they released manufacturer guidance and now we are seeing devices popping up on Ali which have esphome preinstalled. 

Big fan of both of these projects and I like the idea of a "presence" library which could keep this project with the core functionality without project creep and also allow esphome to also benefit. 

I'm not a developer so what do I know  🤷

drahdiwaberl, 2021-12-07:
as others, i'm having multiple esp32s spread around the house and garden already using them as gateways for sensors like these https://esphome.io/components/sensor/xiaomi_ble.html
so i as well would like to avoid setting up additional receivers for what ESPresense does.

reading this thread and thinking about it i'm curious if - conceptually - it wouldn't make more sense to offload the special functions ESPresense has (Kalman filter, fingerprinting,..) to a more capable processor like the server HA is running on.

of course i understand that this would be a completely different approach and might be out of the scope of this project.

DTTerastar, 2021-12-07:
Passive BLE Monitor is actually working on something like that.  I do think it could make some sense.  But it would be even more chatty than the current mqtt setup.

drahdiwaberl, 2021-12-07:
well that's the other question then of course - if the raw/unfiltered data shouldn't be transferred differently (not mqtt).. curious how things will develop with these projects around that topic..

cpainchaud, 2022-01-19:
Someone seems to have found a way to make it happen with ESPhome with a lambda function:
https://gist.github.com/kmdm/c01c0693b1380d826af200a3e9b2a656

DTTerastar, 2022-01-19:
Yes, that does about 2% of what ESPresence does.  It for sure is possible just a lot of work.

cpainchaud, 2022-01-19:
it surely is less than what ESPresence does but many of us rely on sensors/scenarios where it would require very high level of code customization. I don't mind coding, I have my fair share of repos and forks but I do start finding myself time constrained :)

DTTerastar, 2022-01-19:
Instead of putting the 1 Euro filtering and the fingerprinting into a lambda, you could just forward it all to hass and have a component there in python that does all that...  That's what my earlier comment is.

rpatel3001, 2022-01-24:
If anyone is interested, I put together some code for my own use with esphome that borrows some of the filtering scheme from ESPresense. It doesn't do fingerprinting and is only setup for iBeacons, but if you're handy with code you could easily extend it however you like.

https://github.com/rpatel3001/BleDistance

robloh, 2022-02-08:
I would love for this to be available as inside ESPHome as well. I already have multiple ESPHome (ESP32 Ethernet PoE) throughout my place listening mainly to BLE broadcasts, so it seems like a natural fit and not have 2 devices (in each location) sitting next to each other listening to the same BLE broadcasts :)

While the Lambda solution is one way, ESPHome can also be extended with C++ code (either as a custom component, or integrated in that project): https://esphome.io/custom/custom_component.html

neilvangeffen, 2022-02-10:
every light bulb in my house (30) is getting an ESP32 in it running esphome. Would be great if you could combine these projects, as 30 always on hidden receivers should give me some pretty cool data!!!

nmajin, 2022-02-10:
> every light bulb in my house (30) is getting an ESP32 in it running esphome.

What light bulbs do you have that you can flash?

neilvangeffen, 2022-02-10:
> > every light bulb in my house (30) is getting an ESP32 in it running esphome.
> 
> What light bulbs do you have that you can flash?

Just tuya ones with the shit new module, so replacing them with ESP-C3-12F ones

taha-yassine, 2022-02-11:
> > > every light bulb in my house (30) is getting an ESP32 in it running esphome.
> > 
> > 
> > What light bulbs do you have that you can flash?
> 
> Just tuya ones with the shit new module, so replacing them with ESP-C3-12F ones

Do you have a tutorial on how to do that?

jcastro, 2022-02-12:
> If anyone is interested, I put together some code for my own use with esphome that borrows some of the filtering scheme from ESPresense. It doesn't do fingerprinting and is only setup for iBeacons, [...]

Not super handy with code but any plans to add Apple Watch tracking?

rpatel3001, 2022-02-12:
> Not super handy with code but any plans to add Apple Watch tracking?

No, sorry. I don't have one to test with. If someone hacks it in I'd accept a PR to the repo though.

ivdimitro, 2022-02-17:
The biggest benefit I can see of combining the code with the ESPHome is the possibility to flash a ESP based smart plug and use it for tracking as well. 
The problem I have currently is to find suitable enclosure that is attached to wall AC socket. The boxes you shared above are nice but I can't find them for sale in single units.

Guy-Falkes, 2022-03-09:
I'm using ESPHome and a ESP32 to controll my window covers with AM43's installed. It would be awesome if I could also use that ESP32 for ESPresence. That way I would not have to install a different ESP32 for each function ;)

bjeanes, 2022-04-27:
> What light bulbs do you have that you can flash?

https://www.athom.tech/blank-1/15w-color-bulb-for-esphome are some, but they are ESP8285 not ESP32 so not useful for these purposes. 

I have about 10 ESPHome devices, 4-5 of which are ESP32s. I know there are diminishing returns to vocalising support for an ESPHome component that leverages the ESPresense code, but I think the use case really is strong.

ESPHome has been designed as a platform/foundation for modular, orthogonal use-cases to support a wide range of sensors, switches, lights, device control, BLE interaction. Bringing the core feature set of ESPresense as a component for ESPHome will be easier than gradually adding more and more sensor support to ESPresense. Even just modularising the core logic into a re-usable library so that a custom ESPHome component can be made by the community would be a huge leap forward.

CallMeAreks, 2022-05-09:
As others have mentioned, this would be an incredible match. ESP32 are cheap, but I think it's less about the money and more about the what it takes to actually deploy a good network of them. There are new switches like the sonoff switchman which run using ESP32's, so having your light switch double down as a room presence detector would be unmatched in terms of usability.

It would integrate seamlessly with your home without  having to get creative with the placements so it's either not visible or not a complete eyesore : ) Even without considering the sonoff switchman, which might not be for everybody, merging devices into one is simply half the hassle.

I do understand however this might be much more complex than 'let's merge them'. But nevertheless worth to evaluate, IMO.

bjeanes, 2022-05-09:
> new switches like the sonoff switchman which run using ESP32's

Also the Shelly Plus series, like the [Shelly Plus 1](https://shelly.cloud/knowledge-base/devices/shelly-plus-1/) -- which are ESPHome flashable.

Guy-Falkes, 2022-05-09:
> As others have mentioned, this would be an incredible match. ESP32 are cheap, but I think it's less about the money and more about the what it takes to actually deploy a good network of them. There [...]

WOW, i did not even know this was a possibility. Given the availability of a ESP32 switch (which can be flashed with ESPHome) it is so very logical to merge the functions and use that one switch as both switch and for instance room presence detector.

Please, please consider looking into getting ESPresence available through ESPHome (or inform me of another working solution to have the switch with ESPresence connected to HomeAssistant).

RaZer0r, 2022-05-15:
YES to that. I have an sonoff m5 in each room, and run them with esphome ;). If bel tracking can be done from those, this whole room_tracking got a lot more interesting fast :D

fmnamado, 2022-07-08:
Another one for this please.
I already have esphome running in ESP32 around the house.

would love to integrate espresense into the same hardware.

doesn’t make sense having another esp32 just for this

can you pleaseeeee make this happen?

thank you very much for your hard work

DTTerastar, 2022-08-30:
I'm going to close this because I just don't think it'll be possible to do this with 328k of RAM and 4MB of flash.  And if it was done it would be a totally different project.  ESPs aren't little raspberry pis, they have their limits.

closed 2022-08-30 by DTTerastar

rpatel3001, 2022-08-30:
Just going to reup my custom header for esphome that implements the filtering of espresense, which is like 50% of its advantage over using esphome's raw rssi. Someone with an apple watch can hack in the fingerprinting and I'll accept the PR and that'll be the other 50% of its advantage.

https://github.com/rpatel3001/BleDistance

danps1, 2023-01-24:
Hi,

I created a Home Assistant feature request which I think could provide an alternative approach here as well, if you're like to look at https://community.home-assistant.io/t/apple-devices-presence-using-esphome-ble-proxy-espresense-alternative/524019/1

Thanks

Dan

mvgijssel, 2023-01-26:
For POC purposes I implemented bluetooth tracking in ESPHome similar to ESPresence here https://github.com/mvgijssel/setup/pull/128 which includes Apple Watch tracking using the IRK method. This has not the polish that ESPresence has (for which I have a lot of ❤️), but proves that ESPresence tracking functionality can work in combination with ESPHome sensors.

danps1, 2023-01-26:
@mvgijssel - this looks really cool, and I'd love to give it a go, but it looks like I'm doing something wrong when I try...

I add your git repository as an `external_component` like this...
```
external_components:
  - source:
      type: git
      url: https://github.com/mvgijssel/setup/tree/master/esphome_nimble
```
However, maybe I can't use the URL like this because I get this error?...
```
    repository 'https://github.com/mvgijssel/setup/tree/master/esphome_nimble/' not found.
```
Thanks for any pointers

Dan

btw - I also have a lot of ❤️ for ESPresence, but I just don't want to have 2x ESP32 for every room!

mvgijssel, 2023-01-26:
> @mvgijssel - this looks really cool, and I'd love to give it a go, but it looks like I'm doing something wrong when I try...
>
> I add your git repository as an `external_component` like this...
>
> ``` [...]

Seems you would need to set it up like so:

```yaml
external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  - source:
      type: git
      url: https://github.com/mvgijssel/setup
      ref: master
```

but that doesn't work, because I'm not following the documentation as described here https://esphome.io/components/external_components.html#git 🤦 where all esphome (external) components need to be in the `components` or `esphome/components` directory.

For now I guess you can inline the code into your project and use the local external component setup:

```
  - source:
      type: local
      path: esphome_nimble
```

until I've moved the project to a different directory.

kvoth1, 2023-09-02:
https://everythingsmarthome.github.io/everything-presence-one

DrSpaldo, 2024-07-01:
I also would love to have one device that does reliable presence detection and bluetooth proxy. Given the recent improvements in adopting devices in ESPresence, I wonder if it is any easier now?

mihsu81, 2024-07-02:
> I also would love to have one device that does reliable presence detection and bluetooth proxy. Given the recent improvements in adopting devices in ESPresence, I wonder if it is any easier now?

You could try https://github.com/agittins/bermuda, which uses the BLE Proxy feature of ESPHome and Shelly Gen2 and Gen3 devices.