ESPresense/ESPresense #78: Auto Update not working

ee02217, 2021-10-05:
Hi,

Auto update is not working, please see below error:
[image]

I believe the reason is that the assets are missing from the releases:
[image]

Versus latest release with update working:
[image]

DTTerastar, 2021-10-05:
I think GitHub actions are broken atm.  I can't get anything to kick off.  But yes, the release bins not being there is the root cause of auto update not happening.

DTTerastar, 2021-10-05:
GitHub actions all kicked off finally.  🤞