ESPresense/ESPresense #80: Add output GPIO for presence driven action

ivanvach, 2021-10-05:
Add output GPIO for presence driven action such as unlocking a door for example.

Activate the output upon presence detection and deactivate it otherwise. 

Input GPIOs are being offered now for connecting external presence sensors such as PIR sensors. This is very smart and very nice. 
Adding output GPIO will make it even nicer and smarter.

corgan2222, 2021-10-05:
I had this in mind for some time.  
Something like this ?

- define one or more gpios in the settings
- iterate through the given pin(s) 
- mqtt/set to create a switch entity in ha for each gpio
- use the switch(es) in any automation to send a mqtt message to set the gpio pin to 0/1.

DTTerastar, 2021-10-05:
I feel like we're recreating Tasmota/ESPHome in here... Are we sure this is our core competency?

corgan2222, 2021-10-05:
> 
> 
> I feel like we're recreating Tasmota/ESPHome in here... Are we sure this is our core competency?

I had the same thoughts! 
For me, I learn a lot while working on these things and have really fun while implementing new features. Because at this point, these are all things I would use by my own. 
I think important is to be clear in communication, that we are not a tasmota style project and we don't make promises which may could disappoint people in the end.

DTTerastar, 2022-04-04:
I think we should do this in the end because people would like to flash shelly devices and not lose the switch capabilities

ivanvach, 2022-04-04:
Good point.

Appyx, 2022-04-25:
@DTTerastar Does this mean shellys are supported? This would be awesome to have invisible trackers everywhere.

martinisoft1, 2022-04-25:
At first, what a great project this is! many thanks!
For me I think it would be great if we could control the GPIO trough MQTT.
This would enable us to for instance control door locks, or to control proprietary smart locks by controlling a modified key fob trough a voltage free contact from a relay that in turn in connected to the ESP32. The ESP32 could funtion as a complete bridge for proprietary smart locks this way. A nice and cleans solution.
If possible I would also like to be able to control the LED (in my case of the Atom lite) trough MQTT giving us the option to signal the status of this DIY bridge trough the buildin LED.

DTTerastar, 2022-04-29:
I think those are all good ideas.  We need some more programmers to help implement all of it!

DTTerastar, 2022-04-29:
> @DTTerastar Does this mean shellys are supported? Would be awesome to have invisible trackers everywhere.

I have the weird 1.27mm pitch headers on order from ali, but I might just give up and make a cable this weekend.  I too would like the sensors to be stealth in-wall.  Then I won't have to yell at family members not steal my usb port for charging.

closed 2022-11-08 by stale[bot]

bananajoe86, 2022-11-13:
Are there any activities? I found an output gpio more than helpful for controlling via bluetooth presence.
Joe