ESPresense/ESPresense #82: Unique ID for tile trackers

hmjgriffon, 2021-10-06:
I bought a 2020 tile mate, it has lots of random Mac addresses. I'd like to request a unique fingerprint such as exists for apple devices. I am willing to assist in any way I can as a non programmer.

DTTerastar, 2021-10-06:
I have lots of tiles, and they are all unique.  Can you attach a serial log from: https://espresense.com/terminal ?

DTTerastar, 2021-10-06:
And it keeps doing that?  Are you 100% sure you don't have 2 tiles around?  If you put your tile on top of the base station do you see the Close | message?  

Very strange.  I've never seen a tile be randomized.  How long between rollovers?

hmjgriffon, 2021-10-06:
I held it up to the base station and it did do the closed thing. It was working for a bit, then I saw another tile I'd pop up and it stopped updating in home assistant now it shows not home, it shows the Mac I was using del (304s) attached are pics from that software.

.
[image]
[image]

RichieFrame, 2021-10-20:
FYI the tiles actually do have 2 MAC addresses, one is used for setup, the other for tracking. I have also seen people mentioning their tracking MAC has become randomized after a firmware upgrade, but have been unable to replicate it using 2020 Tile Pro models.

DTTerastar, 2021-11-09:
That's definitive proof... That stinks!

sotatech, 2021-11-09:
To add further to that, I switched to a Tile Pro and it works fine for over a week now.

medains, 2021-11-10:
I have a single tile (from a pack of 4) that exhibits this behaviour - changes approx every 15 minutes.
The one that's misbehaving is one of the stick-on tiles.
I might delete and re-add to the app to see if it changes that behaviour.

DTTerastar, 2021-11-11:
Try changing the tune.  That makes it re-upload firmware.

medains, 2021-11-11:
> 
> 
> Try changing the tune. That makes it re-upload firmware.

Thanks for the tip - got continued rotation after changing the tune though :(

If I'm feeling brave, I'll change the tune on another one and see if that triggers the unwanted behaviour.

Zimbu98, 2021-12-26:
I would suggest that you use an iBeacon instead. With an iBeacon, you can easily set a fixed and unique UUID for each beacon using the beacon configuration app. iBeacons also have fixed MACs, so you can use the MAC also, but the "proper" way is to use the iBeacon's UUID. With an iBeacon, using the config app, you can also modify the broadcast interval (how often the beacon broadcasts) and broadcast power. This gives you a lot of extra flexibility to fine tune things and save battery power.

DTTerastar, 2022-03-14:
Has anyone else confirmed they're tiles are using random macs?  I still haven't seen it.

closed 2022-05-20 by stale[bot]