ESPresense/ESPresense #97: Mostly "QryErr" results in logs

OMGTheCloud, 2021-10-17:
**Describe the bug**
I get a lot of `QryErr` in the serial logs.  ESPresense devices are "working", but slow to catch their polling intervals. I've tried both with the default settings and also enabling the Active Scanning in the firmware. I'm primarily testing with an iPhone 13 Pro, and an BlueCharm BC011 beacon. I have the two ESPresense ESP32's set up in different rooms, about 50m apart, and moving between the rooms with my beacons, they do eventually update as being "in" that room, but it can take a few minutes. seems odd.  See screenshot: 
[image]

**Hardware Details**
So far running two ESP32 with current version of ESPresense installed and configured.

OMGTheCloud, 2021-10-18:
Update: there seems to be some interaction with this behavior and the `Query devices for characteristics (helps apple fingerprints uniqueness)` option. With that enabled, I do indeed detect my Apple devices' fingerprint, but I get all those `QryErr` messages. With that option turned off, I see none of that, and my "regular" BlueCharm beacon reacts fast when moving between zones.

DTTerastar, 2021-10-22:
Uptime is MUCH better: 
[image]

Are you seeing any improvement?

DTTerastar, 2021-10-23:
I'm all Unifi w/ the same SSID for 5ghz and 2.4 and I don't see that issue.

But yes, if you're seeing the captive portal popup that means the ESP32 is hating your wifi.