ESPresense/ESPresense #110: Disconnected from MQTT

faizalgazali, 2021-11-02:
I already set up 2 espresence and seems to be working just fine. Trying to add another espresence and its keep disconnecting from MQTT.  I am just a newbie and do not really understand what is going on :)

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (C8:C9:A3:F9:0C:04)...
Connecting to WiFi SSID ASUS2nd_repeater.................192.168.1.108
Version:     v2.0.6
IP address:   192.168.1.108
DNS address:  192.168.1.1
Hostname:     espresense-lvgroom
Room:         Lvgroom
Max Distance: 3.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor: 
Connecting to MQTT 192.168.1.2 1883
0 New   | MAC: da4d8c9fd8a1, ID: md:3412:14                                                   
0 New   | MAC: 60a0a0a04ce2, ID: sid:0xfe9f                                                   
0 New   | MAC: 477ea2e1bb42, ID: sid:0xfe9f                                                   
0 New   | MAC: d94ea0a60f0c, ID: mifit:d94ea0a60f0c                                           
0 New   | MAC: 459a1c7eb6c9, ID: iBeacon:782583e3-a6e5-4ed3-b820-46fca6bbc70f-100-1           
0 New   | MAC: 6836251864ee, ID: apple:1005:9-12                                              011855d03d
[E][main.cpp:202] onMqttDisconnect(): Disconnected from MQTT; reason 0

DTTerastar, 2021-11-03:
Can you enable logging on your mqtt server to see if it's disconnecting it for a reason?  Having the same name for a node can do it or an incorrect login can as well.

faizalgazali, 2021-11-04:
> Can you enable logging on your mqtt server to see if it's disconnecting it for a reason? Having the same name for a node can do it or an incorrect login can as well.

Thank you for your response. I will update once I get back from my holiday next week.

faizalgazali, 2021-11-08:
> Can you enable logging on your mqtt server to see if it's disconnecting it for a reason? Having the same name for a node can do it or an incorrect login can as well.

1636390391: New connection from 172.30.32.2 on port 1883.
1636390391: Socket error on client <unknown>, disconnecting.
1636390421: Client DVES_E8FE4B has exceeded timeout, disconnecting.
1636390511: New connection from 172.30.32.2 on port 1883.
1636390511: Socket error on client <unknown>, disconnecting.
1636390631: New connection from 172.30.32.2 on port 1883.
1636390631: Socket error on client <unknown>, disconnecting.
1636390751: New connection from 172.30.32.2 on port 1883.
1636390751: Socket error on client <unknown>, disconnecting.
1636390871: New connection from 172.30.32.2 on port 1883.
1636390871: Socket error on client <unknown>, disconnecting.
1636390991: New connection from 172.30.32.2 on port 1883.
1636390991: Socket error on client <unknown>, disconnecting.
1636391111: New connection from 172.30.32.2 on port 1883.

DTTerastar, 2021-11-09:
Can you paste your mqtt server config?  Have you been able to use it with any other integrations?

faizalgazali, 2021-11-09:
> Can you paste your mqtt server config? Have you been able to use it with any other integrations?

logins:
  - username: mqttname
    password: P4sswords
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false

Integration that I use with this MQTT are tasmota, zigbee2mqtt and 2 of my other ESPresense (esp32). I just do not understand why I was unable to add more ESPresense (esp32)

DTTerastar, 2021-11-09:
I'd try using the "Clean install" option.  I don't see any reason for it not to work.  A *wireshark* log would be the only way to know at this point...

faizalgazali, 2021-11-09:
> _wireshark_

Yes, I had used the clean install every time. What should I look for in Wireshark?

faizalgazali, 2021-11-10:
Update on my issue, MQTT had detected my all 4 my espresense. I do not know what happend. What I did change was connecting to my deco mesh instead off wifi repeater from xiaomi. All works, thanks for the great job.

DTTerastar, 2021-11-10:
Awesome

closed 2021-11-10 by DTTerastar