ESPresense/ESPresense #116: DTH22

margag02, 2021-11-03:
I have a DTH22 install on the esp32 but it's always report humidity = nan and temperature = nan

**Expected behavior**
it should read the temp and humidity correctly

ESP32, DTH22 and config ESPresence dht22 = 1

[image]

jbishop129, 2021-11-04:
Correct sensor name is “DHT22”. I have many of these working correctly with ESPresence. Suspect this is a wiring issue. Recommend verifying your pin assignment and input voltage. These sensors do not need any extra resistors or anything like that, and work equally well on 3v3 and 5v.

margag02, 2021-11-04:
Thanks, found the problem... i tought the setting for espresence for DHT22 was 0 = disable and 1 = enable.  instead of the number of the pin was DHT22 was assigned.

closed 2021-11-04 by margag02