ESPresense/ESPresense #119: Pixlr Tracker

dstachio, 2021-11-07:
Using Pixlr Tracker, it displays the same ID in MQTT explorer.  It will flash between MAC addresses over and over so unable to use more than 1 tracker device.

I know this may not be  a espresense issue, but thought I would report it.

DTTerastar, 2021-11-08:
Can you try and flash this version: https://github.com/ESPresense/ESPresense/actions/runs/1434740101

I need to add to the web flasher the option to pick a prelease...

dstachio, 2021-11-08:
Flashed onto a test board.

MD:0000:8 does not show on test board.
sid:0x0f3e shows on test board and flashes between the 2 mac addresses
sid:0x0f3e0x1802 shows when i activate the Trackr with the button on it.  If I push the button on a second trackr it than begins to flash between the 2 MACs

Does not look like the second tracker is being 

[image]
[image]

DTTerastar, 2021-11-09:
Try https://github.com/ESPresense/ESPresense/actions/runs/1439025058. I made that sid into a trackr:{mac}. Check that the macs don't randomize.

terryhonn, 2021-11-09:
Came here to file bug on Trackr and found this issue. Thrilled to see it already being worked on. 
I can confirm that the test build does differentiate between different Trackrs, but seem to be missing RAW, DISTANCE, and SPEED.

{"id":"trackr:c8eb03e80008","name":"tkr","rssi@1m":32703,"rssi":-60,"mac":"c8eb03e80008","raw":null,"distance":null,"speed":null}

Glad to help troubleshoot if necessary.

DTTerastar, 2021-11-10:
Ok, i think i have that fixed in: https://github.com/ESPresense/ESPresense/actions/runs/1442648720

terryhonn, 2021-11-10:
> Ok, i think i have that fixed in: https://github.com/ESPresense/ESPresense/actions/runs/1442648720

Thanks for the quick response! I'm getting the following when trying to flash either esp32.bin or esp32-verbose.bin:
Unexpected error: The firmware binary is invalid (magic byte=50, should be E9)

DTTerastar, 2021-11-11:
Can we close this?

terryhonn, 2021-11-11:
Looks great, thank you!  I didn't open, @dstachio did, but it's definitely working for me now. Thanks again!

dstachio, 2021-11-11:
Thanks all. Appreciate the help and fix!

closed 2021-11-11 by dstachio