ESPresense/ESPresense #122: Add ethernet support for ethernet enabled ESP32 devices

joshuakohut77, 2021-11-09:
I want to install my ESP 32 devices around the outside of my house and am worried about ethernet connectivity. 
I found some POE Ethernet based ESP32 modules. I would like to be able to have the option to connect via ethernet instead of WiFi if ethernet is available on the hardware. 

I am will to do any testing if that would be of any help.

DTTerastar, 2021-11-09:
Which module are you going for?  On the ESP every type of ethernet is different :(

joshuakohut77, 2021-11-09:
> Which module are you going for? 
The modules I have are the Olimex ESP32-POE which can be found [here](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware). 
That's frustrating they're all different but I'm not surprised. I just really like the idea of providing power and ethernet based comms with one cable. 
I am willing to try and dig in to get any additional technical information if that would be of use to you. I'm sure you have lots of other things you are working on.

DTTerastar, 2021-11-11:
I actually have a `Olimex ESP32-POE`, I can try and see if I can get it to work.

MarkusFelix, 2021-11-24:
I would also love support for the - well at least when they were new - cheap WT32-ETH01 board.
The pin definitions of those two are as follows:

```c
ethernet_settings ethernetBoards[] = {
   // WT32-EHT01
   {
     1,                 // eth_address, 
     16,                // eth_power, 
     23,                // eth_mdc, 
     18,                // eth_mdio, 
     ETH_PHY_LAN8720,   // eth_type,
     ETH_CLOCK_GPIO0_IN // eth_clk_mode
   },

   // ESP32-POE
   {
      0,                  // eth_address, 
     12,                  // eth_power, 
     23,                  // eth_mdc, 
     18,                  // eth_mdio, 
     ETH_PHY_LAN8720,     // eth_type,
     ETH_CLOCK_GPIO17_OUT // eth_clk_mode
   }
 };
```

I would be very willing to do some beta testing and, if I find the time I'll take a look at integrating those as well.

joshuakohut77, 2021-12-16:
I don't want to rush you or anything. I just wanted to check and see if there has been any updates to this or if there's anything I can do to assist. Thanks for all the great work you have done on this project, it works great.

sktaylortrash, 2022-01-19:
If this is still being considered I'd like to also throw the Lilygo TTGO T-Internet-POE into the mix

ethernet_settings ethernetBoards[] = {
   // T-Internet-POE
   {
      0,                  // eth_address, 
     23,                  // eth_mdc, 
     18,                  // eth_mdio, 
     ETH_PHY_LAN8720,     // eth_type,
     ETH_CLOCK_GPIO17_OUT // eth_clk_mode
   }
 }

bramstroker, 2022-04-01:
Any change this still gets implemented? I also have an "Olimex ESP32-POE". 
This is a really nice and promising project, and got 2 ble nodes tracking correctly with the ESP connected to Wifi.
However at the location where I need the ESP installed I have very weak wifi (because of hundreds of neighbouring SSID's which interfere my Wifi signal). I have 2 Unifi AP's and tried a lot of different settings, but I can't get any reliable wifi on that specific location.
However I have an ethernet connection there so it would be very nice if I could use my POE enabled ESP.

DTTerastar, 2022-04-01:
WLED has a very nice implementation we need to borrow.  Yes I'd still like to support this use case.  It's in my list of 2 or 3 next things to do.

bramstroker, 2022-04-01:
@DTTerastar That's awesome! Thanks for the headsup.

DTTerastar, 2022-06-04:
This was implemented already.

closed 2022-06-04 by DTTerastar