ESPresense/ESPresense #123: Room keeps bouncing between other rooms

iamandymcinnes, 2021-11-10:
Hi

This is possibly a config issue but, I've put an esp32 in each room of my house. I've configured the max distance down to around 5m/6m for most rooms as I've positioned them in the far corners between rooms to maximise the distance from a esp in another room to avoid geting picked up by another esp whose max distance you also satisfy. However with them being on different floors too I am finding particularly in my lounge where my snug is below, that it shows me flip-flopping between the two.

Now I understand the max distance allows each esp to filter results that greater than its max distance, but there is always going to be some overlap I assume, so that would mean that home assistant needs to do something in order to pick the esp with the strongest signal or something?

Is there some configuration to handle this?

Many thanks

Andy

mjcumming, 2021-11-10:
I think the better solution is going to be training a collection of BT sensors system using XGBoost as done in this post https://github.com/mayorquinmachines/retrievr

iamandymcinnes, 2021-11-10:
Is that something on the home assistant side like an addon that will calculate which room is the most appropriate fit?

DTTerastar, 2021-11-11:
https://github.com/ESPresense/ad-espresense-ips is the start of exactly this.  I want to use [XPGBoost](https://xgboost.readthedocs.io/en/latest/) to do it.

iamandymcinnes, 2021-11-11:
@DTTerastar ah nice, yeah I had already set that up, but I'm not enitrely sure if it's working, not sure what I would expect to see from it exactly?

This is the config I have currently for the esp nodes I've configured so far. I set the Snug which is under the lounge at 0, 0, 0.9 as I use that as my starting corner and the sensor is 0.9m off the ground. The Lounge sensor is in the same place, but the floor about so I've set the y to 3.1:

`ESPresenseIps:
  module: espresense-ips
  class: ESPresenseIps
  pluggins:
    - HASS
    - MQTT
  rooms:
    Snug: [0, 0, 0.9]
    Lounge: [0, 0, 3.1]
    DrawingRoom: [0, 9, 3.1]
    Kitchen: [6, 13.5, 3.1]
  devices:
  - id: tile:cd1dcb7ae277
    name: Claire Keys Tile BLE
    timeout: 30
    away_timeout: 120
  - id: tile:e572c53969ee
    name: Andy Keys Tile BLE
    timeout: 30
    away_timeout: 120
  - id: tile:cd1dcb7ae277
    name: Car Keys Tile BLE
    timeout: 30
    away_timeout: 120
  - id: apple:1006:10-12
    name: Andy Claire IPhone BLE
    timeout: 30
    away_timeout: 120`

I AppDeamon I'm not seeing any callbacks happening, I do see it find and register the mqqt queues for each room configured, but not sure what I should see in terms of the output?

iamandymcinnes, 2021-12-01:
Hi 

Just wondered if there was any update on the IPS implementation and/or XPGBoost changes? Happy to help out with any testing or where ever I can!

Many thanks

Andy

DTTerastar, 2021-12-02:
I've been stuck for a bit... Just got AppDaemon working again... I'm pretty frustrated at all the different automation methods of HA.  HA is so great at getting the entities for all the stuff in my house.  But actually making stuff do "smart" things is driving me nuts.  There has got to be a better way.  I spend a lot of time just thinking about how to fix it.

DTTerastar, 2021-12-02:
Just tried to get a list of Areas in app daemon and found it's impossible atm: https://github.com/AppDaemon/appdaemon/issues/505

Custom_components of HA can use numpy, but i'm pretty sure scipy is off limits so that's out.  ANNOYING!  Maybe I'll try node-red...

iamandymcinnes, 2021-12-02:
I'm a dotnet developer, although am not entirely family with the functional requirements required to train a model with positional data and then use that to suggest a room or area you are in.

I've had a quick look at NetDaemon and that has area support, and XGBoost has .net wrapper https://github.com/PicNet/XGBoost.Net

I also looked at the code for the retrivr example: https://www.hackster.io/home_skillet/alexa-where-are-my-keys-aa0e82#code

I haven't written anything bar the hello world example with NetDaemon so not sure if that could be used to replicate this behaviour. I also wasn't sure how you would deal with the learning process in order to collect data to build a model?

As an aside I'm not sure what the limitations of custom_components are, but is it not possible to achieve what you want there?

Happy to try and help either way, but I'd need a bit guidance.

DTTerastar, 2021-12-02:
I'm a dotnet dev as well.  I have tried NetDaemon before but, I was trying to go the python route since it has all the great scipy/numpy/pandas stuff in it native.

iamandymcinnes, 2021-12-08:
I've been reading loads of things trying to find some way of debouncing the ble location of a device but am still at a loss. Like most I've gone the esp32 route as they are cheaper and lower powered than having pi zeros everywhere, and also I tend to already have them in rooms or I had plans to have a multisensor in each room, so they just make sense from a technology perspective. I've been trying to workout if things like monitor are any benefit or if they can even handle the scenario we have where espresence is publishing the devices it discovers.

I don't know enough about the restrictions of running python alongside home assistants core and my experience of all that pyenv stuff are limited to where I've had to dabble not out of choice. Maybe a slightly sledge hammer approach, but also possibly an interesting approach.... What about just creating an addon using a python base image, or one of the ha standard bases. That way you can get what you want and still integrate with HA. I can probably help a little with this as I've dabbled with a couple of addons. The only thing that I would love to see but haven't managed yet with this is getting the addons output as an integration in HA. But you can probably just create a HACs custom component to provide that functionality, which seems to be the way others achieve the same thing.

DTTerastar, 2022-03-22:
I think this should be closed since this is on the core firmware which is working as designed.  We just need something else higher level to smooth things out and classify the room based on multiple sensors.

ffirenmatana, 2022-04-15:
Not a lot to add except this behaviour is far less noticeable on Apple based devices. In fact I've reliably got this working using MacBooks and iPads and 5 sensors, with or without customising the calibration. 

Anything Android (I only have Samsung to test) exhibits the bouncing behaviour to the point of using it unusable for geofencing. Even if placed adjacent to a sensor, it will still hop to a node 12M away.

DTTerastar, 2022-04-17:
Hmmm, that is interesting.  I thought iBeacons from HA Companion would be BETTER not worse.  What is the interval you're seeing?  (It's a json property of the detected iBeacon).  One thing I've noticed is the GUI of HA Companion is wonky, sometimes you need to change the settings 2 times to get them to stick.  You want the Advertise mode set to low latency in order to have enough beacons to average the rssi.

ffirenmatana, 2022-04-17:
You were quite correct. The json was disparate to the settings and re-applying the Advertise mode fixed it. Works perfectly for me now. Thanks mate

DTTerastar, 2022-04-24:
Awesome!

closed 2022-04-24 by DTTerastar

ozett, 2023-01-03:
maybe for the records here: 

use kalman filter on RSSI ?

https://github.com/1technophile/OpenMQTTGateway/issues/783