ESPresense/ESPresense #125: No captive portal after flashing new board from the web installer

brgerig, 2021-11-11:
**Describe the bug**
I used the web installer to flash new ESP32 boards (tried with the "clean install" option both checked and unchecked). They create an SSID to connect to, but I'm not able to load the captive portal to configure them. Directly accessing http://192.168.4.1 doesn't work either.

**Expected behavior**
Connect to AP-mode network (espresense-xxxxxx). Get redirected to configuration portal.

**Logs**
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (58:BF:25:83:B1:30)...
I (160) wifi:wifi driver task: 3ffd42dc, prio:23, stack:3584, core=0
I (543) wifi:wifi firmware version: dc30037
I (544) wifi:config NVS flash: enabled
I (544) wifi:config nano formating: disabled
I (544) wifi:Init data frame dynamic rx buffer num: 32
I (549) wifi:Init management frame dynamic rx buffer num: 32
I (554) wifi:Init management short buffer num: 32
I (559) wifi:Init dynamic tx buffer num: 32
I (563) wifi:Init static rx buffer size: 1600
I (567) wifi:Init static rx buffer num: 16
I (570) wifi:Init dynamic rx buffer num: 32
I (667) wifi:mode : sta (58:bf:25:83:b1:30)
First contact!

I (680) wifi:flush txq
I (680) wifi:stop sw txq
I (682) wifi:lmac stop hw txq
Starting access point for configuration portal.
SSID: 'espresense-30b183'
I (695) wifi:mode : softAP (58:bf:25:83:b1:31)
I (696) wifi:Total power save buffer number: 16
I (696) wifi:Init max length of beacon: 752/752
I (697) wifi:Init max length of beacon: 752/752
192.168.4.1
I (17301) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (17303) wifi:station: 4a:40:35:a8:9f:c7 join, AID=1, bgn, 20
dhcps: send_nak>>udp_sendto result 0
```

**Hardware Details**
ESP32 dev kit board, ESP-WROOM-32 

**Screenshots**
None

DTTerastar, 2021-11-11:
Do you see an IP address assigned? What phone/os?

brgerig, 2021-11-11:
> The send_nak seems bad. This kinda relates: [espressif/arduino-esp32#2285](https://github.com/espressif/arduino-esp32/issues/2285)

I think the send_nak may have been from my computer switching networks and trying to use its old address before getting a new one.

> Do you see an IP address assigned? What phone/os?

Happens with both Windows 10 and my Samsung Galaxy S21 running Android 11.

I do get an IP address:
```
Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Intel(R) Dual Band Wireless-AC 8265
   Physical Address. . . . . . . . . : -----------------------
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::ac12:4a58:5e22:c116%10(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.168.4.2(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Lease Obtained. . . . . . . . . . : Thursday, November 11, 2021 6:30:32 PM
   Lease Expires . . . . . . . . . . : Thursday, November 11, 2021 8:30:32 PM
   Default Gateway . . . . . . . . . : 192.168.4.1
   DHCP Server . . . . . . . . . . . : 192.168.4.1
   DHCPv6 IAID . . . . . . . . . . . : 156268207
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-24-07-41-E5-C4-65-16-06-B8-75
   DNS Servers . . . . . . . . . . . : 192.168.4.1
   NetBIOS over Tcpip. . . . . . . . : Disabled
```

My computer is able to detect that it's a captive portaled network and opens up a browser tab to http://www.msftconnecttest.com/redirect. However, it doesn't go anywhere. Once I'm connected, though, nothing goes anywhere and DNS is unresponsive:
```
C:\>nslookup
DNS request timed out.
    timeout was 2 seconds.
Default Server:  UnKnown
Address:  192.168.4.1

> msftconnecttest.com
Server:  UnKnown
Address:  192.168.4.1

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
*** Request to UnKnown timed-out
> clients3.google.com
Server:  UnKnown
Address:  192.168.4.1

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
*** Request to UnKnown timed-out
```

brgerig, 2021-11-12:
Well, I'm not sure what happened, but eventually after multiple reboots and retries, I was able to access the web interface on one of them. I'll certainly try the other one at some point and see if it starts cooperating, too.

DTTerastar, 2021-11-12:
I added a log print to v2.0.11 as well as a tweak to the timing.  Try it and see if it works better.  I tried on my m5atom and it worked great first try.

brgerig, 2021-11-13:
That version did seem to work more consistently. Although there was still no automatic redirect to the right URL, I was able to go to the IP manually without much issue. Thanks!

closed 2021-11-13 by brgerig