ESPresense/ESPresense #131: Suited beacon, MiBand 6 experience

ThomDietrich, 2021-11-18:
Hey all,
among all the possible beacons, I feel the MiBand best fits my use case (arriving/leaving home). Are you aware whether the MiBand 6 (or the 5) works in a recommended way (e.g. quick update frequency)? Should I go for a Tile Mate (2022) instead?

I intend to buy one of those and would be happy to report my experience for the table at https://espresense.com/beacons
I love that you provide this table but for a potential user like myself, more details would be great! Happy to do my part.

Thanks!

DTTerastar, 2021-11-18:
Tiles are very slow.  I don't have a miband, but previous PRs added support for them so if it's miband vs tile, def go with miband.

DTTerastar, 2021-11-18:
Blue Charm beacons are 100% configurable for update speed, so also consider those.

ThomDietrich, 2021-11-18:
Thanks! I suggest to keep this ticket open to collect these experiences, then document them on the webpage

jrb80, 2021-11-26:
I am using the following low cost fitness bands/watches with good success;
- [Huawei Band 4 Pro-AC0](https://consumer.huawei.com/uk/wearables/band4-pro/)
- [VeryFitPro ID205](https://www.aliexpress.com/w/wholesale-id205-smart-watch.html) (Shenzhen DO Intelligent Technology)

The [VeryFitPro fitness bands ](https://www.idoosmart.com/products.html )come under a hundred different marketing names but if you search the model numbers in AliExpress/Amazon/eBay you will find many variants, all for a very low cost.

Personally, I would like support for the [Samsung SmartTag](https://www.samsung.com/uk/mobile-accessories/galaxy-smarttag-black-ei-t5300bbegeu/) which has an incredible read range (compared to AirTag) Samsung randomise the MAC address and it changes every 5 to 10 minutes. It actually works somehow if there is only one tag as ESPresense seems to ignore the MAC but when you introduce a second tag the identifier is duplicated and then ESPresense gets confused. It would be wonderful if a "fingerprint" concept was created for this tag but I realise I could be in a minority here!

DTTerastar, 2021-11-27:
I just setup a Trackr as a dog collar tracker... It's very small, tracks quick, and you can provision it w/ nRF Connect since the app is dead...

jrb80, 2021-11-29:
I am using a [Samsung SmartTag](https://www.samsung.com/uk/mobile-accessories/galaxy-smarttag-black-ei-t5300bbegeu/) for my dog mainly for position relay over Galaxy phones similar to the Airtags, but as I said above ESPresense is unable to distinguish multiple tags as they get reported under the same ID. The [Samsung SmartThings Find](https://news.samsung.com/global/samsung-launches-smartthings-find-a-new-way-to-quickly-and-easily-locate-your-galaxy-devices) app works fairly well in urban environments to aggregate enough position data from nearby phones.  It would be nice if a fingerprint was added in ESPresense somehow. I read that Trackr is pretty much dead now and doesn't offer community based position since the withdrawal of apps, so it's not really an option....

[image]

DTTerastar, 2021-11-29:
Can you investigate if it has a readable id via nRFConnect?

jrb80, 2021-11-29:
> Can you investigate if it has a readable id via nRFConnect?

I briefly looked at it a few days ago, it does connect and read the UUID data, whether that's useful or not. Below is two screens from different tags. I struggled a bit due to the number of BLE devices in my noisy environment (London) and with the MAC changing all the time as the nRFConnect sorts devices by MAC address, so the same devices appears more than once!

[image]
[image]

DTTerastar, 2021-11-29:
0xFD5A def looks like the service uuid for the tag.  Can you scan that after a couple hours and see if you still get 0x15Df..EFAB for it?

jrb80, 2021-11-29:
The "0xfd5a" UUID is constant for both SmartTags, however the data string "0x15D6xxxx" which follows randomly changes. The current behaviour in ESPresense is that all tags publish under the same MQTT ID topic as JSON sample below. 

`{
  "id": "sd:0xfd5a",
  "name": "Smart Tag",
  "idType": 7,
  "rssi@1m": -71,
  "rssi": -75,
  "mac": "1a7e7c60edd6",
  "raw": 1.3,
  "distance": 1.45,
  "speed": 0
}
`

DTTerastar, 2021-11-30:
> { "id": "sd:0xfd5a", "name": "Smart Tag", "idType": 7, ***@***.***": -71,

I can add an "smartTag:" but we don't want to include anything that's
random in the fingerprint.  Can you put a list of the various 0x15D...
data?  If we can find a few bytes that don't change we might be able to
make a consistent fingerprint.

jrb80, 2021-11-30:
> Can you put a list of the various 0x15D data?

Yes of course but is there an easy way to collect the data? It would be good if the serial terminal output printed the UUID Service Data or have it published to the MQTT device topic. Using the nFR Connect app is painful if wanting to collect a large number of data samples for analysis.

jrb80, 2021-11-30:
> Can you put a list of the various 0x15D...

Yes of course but is there an easy way to collect or log the data? It would be good if the serial terminal output printed the UUID service data or have it published to the MQTT device topic. The nFR Connect app is painfully difficult to capture data samples for analysis. I only have a couple of tags so not sure how realible the data could be, though it seems only the prefix 0x15D is constant.

liquidphire, 2021-12-24:
Hi!
I am running into this same issue where my two Smart Tags are basically being ID'd as the same device by ESPresense (sd:0xfd5a), per the above user. 

Wondering if anyone has worked out a workaround for Samsung Smart Tags, or if I should just get some kind of different beacon solution... Happy to contribute data if needed.

Zimbu98, 2021-12-26:
FYI, some of you are confusing key finder tags with beacons. They are both BLE-based, but different in how they function. 

A Tile, for example, is NOT a beacon. A Tile is a BLE tag that requires an ongoing connection to your phone. When the distance becomes too great, the Tile signal will become weak, and then the app will put up a notification e.g. "someone stole your laptop". (lol)

A beacon, on the other hand, works independently from your phone or scanner. It does NOT require a connection, and in fact, stops working when you connect to it. Don't connect to a beacon unless you are configuring it, and you will probably only do that once. 

If you are trying to use an ESP32 as a beacon scanner for presence, then it makes much more sense to use a beacon (aka iBeacon). The beacon with broadcast its unique info package (UUID, Major, Minor, etc.) then your ESP will scan for all beacons in range, see your target beacon, compare the UUID of your beacon to a list of target UUIDs (in your code or stored in the ESP SPIFFS), and then take some action according to how you have coded it. 

Trying to use Tiles or AirTags or FitBit bands will be difficult to use in the same way since they require a connection; you can do it, but it drains the battery and is not needed if you use a real beacon. The bigger problem is that these non-beacon items **don't broadcast any stable ongoing unique identifier**. There are a few hacky ways to make this work, but as far as I know, they require making a quick connection, querying a service UUID, and getting back some stable number such as the product number, or firmware version or whatever. Of course, this doesn't work well if you want to use more than one Tile since these numbers are all identical for all Tiles and can't be changed.

Stooovie, 2022-02-09:
I'm trying to use MiBand 4 and the update is so infrequent it's essentially useless. It's minutes before any changes registers.

bienvangm, 2022-03-01:
I use my MiBand 6 and it could connect, but it only connect when not connected to my phone. If MiBand 6 connected to my phone, I can't track it. Any solution?

antonpopov, 2022-03-25:
> I use my MiBand 6 and it could connect, but it only connect when not connected to my phone. If MiBand 6 connected to my phone, I can't track it. Any solution?

You need to make sure discoverable is turned on within the Mi Fit App.

onlyoneme, 2022-05-28:
I'm testing [MOKO W2](https://www.mokosmart.com/wp-content/uploads/2019/09/Beacon-W2.pdf) wristband actually and it seems to work really nice as a wearable beacon. It has 6 slots which can be configured independently for broadcasting different iBeacon / Eddystone messages simultaneously. Moreover it has an accelerometer which can be used to trigger motion detected advertising with higher frequency for example.

ozett, 2022-05-29:
> I'm testing [MOKO W2](https://www.mokosmart.com/wp-content/uploads/2019/09/Beacon-W2.pdf) wristband actually and it seems to work really nice 

great news. where to buy (international/europe) ?

onlyoneme, 2022-05-29:
> great news. where to buy (international/europe) ?

www.beaconzone.co.uk is the only one European store I'm aware of. And it's only for B2B and quite expensive. Apart of that I've seen those devices only on alibaba.

Zimbu98, 2022-05-29:
Blue Charm sells a couple beacons with motion sensor features but no wristband style. The motion sensor beacons work very well if you can accept the beacon form factor.

rumbu13, 2022-06-20:
Just discovered randomly by watching in mqtt explorer the BLE devices nearby that an old wireless headset (Skulcandy Sesh Evo) is incredibily accurate in communicating its position. Surprisingly, it is identified by espresense as tile:{some mac}.

As long as it's not connected to a phone or something, one of the speakers can safely be kept in a pocket to identify your position around the house :)

closed 2022-09-07 by stale[bot]