ESPresense/ESPresense #132: No captive port wifi found to access configuration after web install

bouton, 2021-11-21:
I am a complete beginner - but a well seasoned hacker in other areas.  I did a web firmware install via chrome on a desktop PC onto a esp32 dev board plugged into a usb port  which was recognized as COM5.  Went apparently ok, no errors, "connecting to wifi" and "First Contact" with an SSID. Then I tried to use my phone to access that SSID (as shown on the video on your page) - but the SSID doesn't come up on my android wifi list of available points. Nor does going to 192.168.4.1 directly via a browser (either on my phone nor on the PC) work.
I am obviously missing something easy peasy, but am lost as to how to proceed to get this to now connect to my network. Looking forward to playing further. Thanks.

mjcumming, 2021-11-21:
Leave the esp32 plugged in to the usb port and go to the terminal page to see the output from the esp32

bouton, 2021-11-22:
Yep. I see the output on the terminal. That is where I got the SSID and IP address. Now I want to sign on to the captive port SSID I got from the terminal, but it does not show up as an available wifi when I go to my pc or phone wifi point options.

bouton, 2021-11-22:
still no luck. when i boot esp32 I get
Using 'COM5' as serial port.
Showing logs:
[23:03:40]ets Jun  8 2016 00:22:57
[23:03:40]
[23:03:40]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:03:40]configsip: 0, SPIWP:0xee
[23:03:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:03:40]mode:DIO, clock div:2
[23:03:40]load:0x3fff0018,len:4
[23:03:40]load:0x3fff001c,len:1044
[23:03:40]load:0x40078000,len:8896
[23:03:40]load:0x40080400,len:5828
[23:03:40]entry 0x400806ac
[23:03:41]Connecting to WiFi (a mac address here)...
[23:03:41]I (157) wifi:wifi driver task: 3ffd42e8, prio:23, stack:3584, core=0
[23:03:41]I (539) wifi:wifi firmware version: dc30037
[23:03:41]I (539) wifi:config NVS flash: enabled
[23:03:41]I (539) wifi:config nano formating: disabled
[23:03:41]I (540) wifi:Init data frame dynamic rx buffer num: 32
[23:03:41]I (544) wifi:Init management frame dynamic rx buffer num: 32
[23:03:41]I (550) wifi:Init management short buffer num: 32
[23:03:41]I (554) wifi:Init dynamic tx buffer num: 32
[23:03:41]I (558) wifi:Init static rx buffer size: 1600
[23:03:41]I (562) wifi:Init static rx buffer num: 16
[23:03:41]I (566) wifi:Init dynamic rx buffer num: 32
[23:03:41]I (660) wifi:mode : sta (a mac address here)
[23:03:41]First contact!
[23:03:41]
[23:03:41]I (673) wifi:flush txq
[23:03:41]I (673) wifi:stop sw txq
[23:03:41]I (675) wifi:lmac stop hw txq
[23:03:41]I (687) wifi:mode : softAP (a mac address here)
[23:03:41]I (688) wifi:Total power save buffer number: 16
[23:03:41]I (689) wifi:Init max length of beacon: 752/752
[23:03:41]I (690) wifi:Init max length of beacon: 752/752
[23:03:41]Starting access point for configuration portal.
[23:03:41]SSID: 'espresense-b439da'
[23:03:42]IP: 192.168.4.1

But no espresense-b439da show up in my phone/pc settings under wifi. Not can I connect to 192.168.4.1. How to proceed.

bouton, 2021-11-22:
I just bought another esp32 - and that worked exactly as describe. connected, visible SSID and see in MQTT broker -
now how do I get the other one to work?

DTTerastar, 2021-11-24:
It looks like you're using the verbose flavor. Have you tried with the non-verbose flavor of firmware?

closed 2021-11-30 by bouton