ESPresense/ESPresense #133: Unable to connect to configured WiFi SSID

circa1741, 2021-11-24:
**Describe the bug**
I am using the MH-ET LIVE ESP32 DEVKIT and I am unable to connect to the configured WiFi. The SSID and password are correctly configured but I do not see where to configure the security type of WPA2-Personal.

**Expected behavior**
Should be able to connect to the configured WiFi SSID.

**Logs**
Connected 115200 8N1...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (FC:F5:C4:14:D9:94)...
Connecting to WiFi SSID Spyglass................................................................................................................................................................................................................................................................................................................................................................................................................ failed.
Starting access point for configuration portal.
SSID: 'espresense-study'
IP: 192.168.4.1

**Hardware Details**
MH-ET LIVE ESP32 DEVKIT
http://esp32.net/images/MH-ET-LIVE/ESP32-DevKit/MH-ET-LIVE_ESP32-DevKit_PhotoTopPerspective.jpg
http://esp32.net/images/MH-ET-LIVE/ESP32-DevKit/MH-ET-LIVE_ESP32-DevKit_PhotoBottom.jpg

**Screenshots**
If applicable, add screenshots to help explain your problem.

DTTerastar, 2021-11-24:
It appears you somehow got your SSID in there with a Carriage Return at the end.  Can you try and configure it again?

circa1741, 2021-11-24:
I configured it again by selecting the SSID from the dropdown. I cannot manually type in the SSID. Still unable to connect to the WiFi.

DTTerastar, 2021-11-24:
Can you try w/ ESPHome, or Tasmota and see if it works any better?  The code I use it pretty bog standard so I can't imagine why you'd have issues.  The ESP32 does only have 2.4ghz radio and no 5ghz support, so make sure your wifi supports 2.4ghz

circa1741, 2021-11-24:
Thanks! Your comment just gave me a hint and reminded me that I I use MAC Address filtering in my router. It is connecting now.

closed 2021-11-24 by circa1741