ESPresense/ESPresense #134: Not working on esp32-c3s

cephos, 2021-11-25:
**Describe the bug**
Latest Binary doesn't run on esp32-c3s. Flashing works but when booting there is always the following repeated error message:
ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x10 (TGOWDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
Invalid chip_id, Expected 5 read 0. Bootloader for wrong chip?
ets_main.c 333
..

**Expected behavior**
ESPresence boots also on this chip.

**Hardware Details**
ESP32-c3s Development-Board

**Compile on Platform fails**
As this chip seems to use another architecture than other esp32 I tried to compile it myself using platformio in vscode. Sadly it fails and I'm not as skilled in C(++) to be able to fix the errors:

> Executing task in folder ESPresense: platformio run <

Processing esp32c3 (platform: espressif32; framework: arduino; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.399f4ec
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-riscv-esp 1.80400.0 (8.4.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftFilters> 0.1.0
|-- <AsyncMqttClient> 0.9.0
|   |-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.18.5
|-- <ESP-WiFiSettings> 3.8.0+sha.9a45666
|   |-- <WebServer> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 2.0.0
|   |-- <LittleFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <NimBLE-Arduino> 1.3.1+sha.245eefc
|-- <AsyncTCP> 1.1.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 2.0.0
|-- <Adafruit Unified Sensor> 1.1.4
|-- <DHT sensor library for ESPx> 1.18.0
|-- <hp_BH1750> 1.0.0
|   |-- <Wire> 2.0.0
|-- <BleFingerprint>
|   |-- <NimBLE-Arduino> 1.3.1+sha.245eefc
|   |-- <ArduinoJson> 6.18.5
|   |-- <SoftFilters> 0.1.0
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <HTTPClient> 2.0.0
[... 148 lines cut ...]
src/main.cpp:39:27: error: 'FIRMWARE' was not declared in this scope
     tele["firm"] = String(FIRMWARE);
                           ^~~~~~~~
src/main.cpp:39:27: note: suggested alternative: 'FMARK'
     tele["firm"] = String(FIRMWARE);
                           ^~~~~~~~
                           FMARK
src/main.cpp: In function 'void connectToWifi()':
src/main.cpp:148:45: error: 'kebabify' was not declared in this scope
     WiFiSettings.hostname = "espresense-" + kebabify(room);
                                             ^~~~~~~~
src/main.cpp:185:10: error: 'slugify' was not declared in this scope
     id = slugify(room);
          ^~~~~~~
Compiling .pio/build/esp32c3/libc74/AsyncMqttClient/AsyncMqttClient/Packets/Out/Connect.cpp.o
*** [.pio/build/esp32c3/src/main.cpp.o] Error 1
================================================================================= [FAILED] Took 9.38 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32c3        FAILED    00:00:09.375
=========================================================================== 1 failed, 0 succeeded in 00:00:09.375 ===========================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

cephos, 2021-11-25:
I forgot, i added the following env to the platformio.ini and tried to build with it:

[env:esp32c3]
platform = espressif32
platform_packages =
	toolchain-riscv-esp
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#master
	platformio/tool-esptoolpy @ ~1.30100
framework = arduino
; take usual esp32dev board and modify meta-info
; to make it a ESP32C3 board.
board = esp32dev
lib_deps = ${common_env_data.lib_deps_external}
board_build.partitions = partitions_singleapp.csv
board_build.mcu = esp32c3
board_build.variant = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.arduino.ldscript = esp32c3_out.ld
; remove old build flags
build_unflags =
  -DARDUINO_ESP32_DEV
  -DARDUINO_VARIANT="esp32"
; inject new ones 
; uncommont Serial macro below if serial is not working..
build_flags =
  -DARDUINO_ESP32C3_DEV
  -DARDUINO_VARIANT="esp32c3"
;  -DARDUINO_SERIAL_PORT=1
;  -DBOARD_HAS_PSRAM
; -DCORE_DEBUG_LEVEL=5

DTTerastar, 2021-11-26:
Yeah C3's are totally different.  RiscV instead of Tensilica.  Will take a while for all the dependent libs to support it.

helix-de, 2022-02-15:
Hello, is there any news here yet? Have here some m5stamp-c3 lying around, which would make themselves, I think proma as a base station. unfortunately I run into the same error as above.

to test the tasmota32-c3 bluetooth firmware flashed works. but no success with esprecense

kirel, 2022-03-04:
Also subscribing to get notified. M5Stamp C3 looks perfect for ESPresence.

DTTerastar, 2022-03-04:
I have one and it's on my list of things to do.  We should setup one of those "what should I work on next" voting boards.

DTTerastar, 2022-03-04:
Bug me in another week if I don't get to it :)

PeterBurner, 2022-03-05:
It would be interesting to see how long the battery live is with something like the [LILYGO TTGO T-OI PLUS](http://www.lilygo.cn/prod_view.aspx?Id=1361)

kirel, 2022-03-28:
Bugging you again ^^

ErnestStaug, 2022-05-11:
No support yet?!

DTTerastar, 2022-05-14:
Sorry, been busy on other things...

idogrf, 2022-05-30:
Support for the c3 would be great!

DTTerastar, 2022-06-04:
Ok, i'm on this next.

DTTerastar, 2022-06-11:
I have something that kinda works!  It's not crashing but it is warning about:

```
E (398825) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (398825) task_wdt:  - IDLE (CPU 0)
E (398825) task_wdt: Tasks currently running:
E (398825) task_wdt: CPU 0: reportTask
```

It warns about basically every task so I'm not sure how to proceed.  The c3 is a very different animal and just barely got supported by the platforms...  It's also showing some pretty crazy rssi swings. Like -37dBm to -100dBm.

closed 2022-06-11 by DTTerastar

idogrf, 2022-06-11:
Thanks @DTTerastar! I'll give it a try...
Is it possible to install the firmware through the browser automatic installer?

DTTerastar, 2022-06-12:
I'll add that now along w/ the option to install betas...

idogrf, 2022-06-12:
Installed it using the [esp32c3.bin](https://github.com/ESPresense/ESPresense/releases/download/v2.2.0b0/esp32c3.bin) file from [v2.2.0b0](https://github.com/ESPresense/ESPresense/releases/tag/v2.2.0b0)
I'm getting non stop `invalid header` errors after flashing and can't find wifi network of the esp
[image]

DTTerastar, 2022-06-16:
https://espresense.com/beta is up, doesn't flash for me correctly either.  Must be missing something.

DTTerastar, 2022-06-16:
My c3 is junk.  Can't read the flash past 2MB.  Gotta order another one...

DTTerastar, 2022-06-16:
Well I still got some results from flashing both good and bad and comparing and it's just 1 byte that is different:

[image]

DTTerastar, 2022-06-16:
Ok patched that 1 byte in the bootloader and now the web flasher works!

[image]

idogrf, 2022-06-17:
@DTTerastar also works for me :)
[image]
I'm never getting past the `starting access point` though... can't see the ssid from any other device

idogrf, 2022-06-17:
@DTTerastar also managed to flash through the web flasher! thanks!
it actually seems to be working pretty good! I get a relative stable rssi reading!

only issue I see at the moment, sometimes it won't connect to wifi when booting up... 
it takes 3 or 4 restarts for it to connect again... any idea?

DTTerastar, 2022-06-18:
I ordered a 5 pack of c3 stamps from digikey so I'll have a known good to test with

TTDARK, 2022-07-26:
> Installed it using the [esp32c3.bin](https://github.com/ESPresense/ESPresense/releases/download/v2.2.0b0/esp32c3.bin) file from [...]

I have the same issue flashing [v2.2.8b2](https://github.com/ESPresense/ESPresense/releases/tag/v2.2.8b2) onto ESP-C3-12F with ESP-Home Flasher. 
Also tried the webflasher on the beta website. It told me the flash was successful, but there was no hotspot to connect to.

idogrf, 2022-07-26:
@TTDARK facing the same issue with no hotspot to connect to after flashing the beta from the website.
Talked to @DTTerastar over on discord and he suggested restarting the esp several times until you can connect to it. 
That didn't work for me but maybe it'll work for you

TTDARK, 2022-07-26:
@idogrf Thanks for your help! It actually worked!
Replugging the power didn't work, but resetting using the webflasher did the job. (ESP Home Flasher always crashes maybe because of so many error messages)
It took about 30s to see the hotspot in wifi analyzer, and another 30s to be discovered by my computer.

idogrf, 2022-07-26:
@TTDARK happy to hear that!
when you say reset using the webflasher, what do you mean? reinstalling the firmware?

shane64, 2022-08-20:
Adding my experience.  Same invalid header issues as described above when using the ESPHome-flasher app on macOS.  Downloaded source code (3.0.0b6) and uploaded to board from PlatformIO which seemed to work fine.  Now get the following on the serial console:

```
01:36:28.923 > ESP-ROM:esp32c3-api1-20210207
01:36:28.932 > Build:Feb  7 2021
01:36:28.933 > rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
01:36:28.938 > SPIWP:0xee
01:36:28.938 > mode:DIO, clock div:1
01:36:28.941 > load:0x3fcd6100,len:0xf4
01:36:28.952 > load:0x403ce000,len:0x8a0
01:36:28.952 > => 0x403ce000: ?? ??:0
01:36:28.952 > load:0x403d0000,len:0x218c
01:36:28.960 > => 0x403d0000: ?? ??:0
01:36:28.960 > SHA-256 comparison failed:
01:36:28.960 > Calculated: 402ae7f8d96ee2a56a5a78dd380569c46de7b91df4dfdb33de30b80f17413178
01:36:28.960 > Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
01:36:28.963 > Attempting to boot anyway...
01:36:28.975 > entry 0x403ce000
01:36:28.975 > => 0x403ce000: ?? ??:0
01:36:28.990 > Setup network
01:36:28.998 > First contact!
01:36:28.998 > 
01:36:28.998 > Starting access point for configuration portal.
01:36:29.004 > SSID: 'espresense-9c4bd5'
01:36:29.004 > Failed to start access point!
01:36:29.498 > IP: 0.0.0.0
```

Neither hard power-off/on nor reset button seems to help.  I've downloaded sample WiFi projects to the board and the Wi-Fi on the board works just fine (can connect to it).  This project doesn't.  

When using source code from 2.2.9, similar behaviour on the console:

```
01:47:41.236 > ESP-ROM:esp32c3-api1-20210207
01:47:41.245 > Build:Feb  7 2021
01:47:41.246 > rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
01:47:41.251 > SPIWP:0xee
01:47:41.251 > mode:DIO, clock div:1
01:47:41.254 > load:0x3fcd6100,len:0xf4
01:47:41.264 > load:0x403ce000,len:0x898
01:47:41.264 > => 0x403ce000: ?? ??:0
01:47:41.264 > load:0x403d0000,len:0x2180
01:47:41.279 > => 0x403d0000: ?? ??:0
01:47:41.279 > entry 0x403ce000
01:47:41.279 > => 0x403ce000: ?? ??:0
01:47:41.284 > Setup network
01:47:41.289 > First contact!
01:47:41.289 > 
01:47:41.289 > Starting access point for configuration portal.
01:47:41.295 > SSID: 'espresense-9c4bd5'
01:47:41.295 > Failed to start access point!
01:47:41.790 > IP: 0.0.0.0
```

Thoughts?  

Thanks.

DTTerastar, 2022-08-22:
The failed to start access point is a clue, we need to figure out "why".  Can you try to flash it via vscode/platformio with the CORE_DEBUG set to 3 or 4?

DTTerastar, 2022-08-22:
Uncomment this: https://github.com/ESPresense/ESPresense/blob/master/platformio.ini#L24

shane64, 2022-08-29:
My apologies for the delay.  Code debug set to level 4 and that line (24) was already uncommented.  Serial monitor shows the following:

```
23:08:14.497 > ESP-ROM:esp32c3-api1-20210207
23:08:14.506 > Build:Feb  7 2021
23:08:14.506 > rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
23:08:14.512 > SPIWP:0xee
23:08:14.512 > mode:DIO, clock div:1
23:08:14.515 > load:0x3fcd6100,len:0xf4
23:08:14.525 > load:0x403ce000,len:0x8a0
23:08:14.525 > => 0x403ce000: ?? ??:0
23:08:14.525 > load:0x403d0000,len:0x218c
23:08:14.533 > => 0x403d0000: ?? ??:0
23:08:14.533 > SHA-256 comparison failed:
23:08:14.533 > Calculated: 402ae7f8d96ee2a56a5a78dd380569c46de7b91df4dfdb33de30b80f17413178
23:08:14.533 > Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
23:08:14.537 > Attempting to boot anyway...
23:08:14.549 > entry 0x403ce000
23:08:14.549 > => 0x403ce000: ?? ??:0
23:08:14.564 > [   6[    69][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
23:08:14.569 > [    69][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
23:08:14.575 > Setup network
23:08:14.578 > [    81][E][vfs_api.cpp:24] open(): File system is not mounted
23:08:14.583 > [    82][E][vfs_api.cpp:24] open(): File system is not mounted
```
last line repeats many more times
```
23:08:14.872 > [   367][E][vfs_api.cpp:24] open(): File system is not mounted
23:08:14.878 > [   372][E][esp32-hal-i2c.c:87] i2cInit(): i2c_param_config failed
23:08:14.883 > [   378][E][vfs_api.cpp:24] open(): File system is not mounted
```
last line repeats many more times
```
23:08:14.972 > [   467][E][vfs_api.cpp:24] open(): File system is not mounted
23:08:14.978 > [   474][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
23:08:14.983 > [   479][E][vfs_api.cpp:24] open(): File system is not mounted
23:08:14.989 > [   484][E][vfs_api.cpp:24] open(): File system is not mounted
23:08:14.994 > First contact!
23:08:14.997 > 
23:08:14.997 > [   500][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
23:08:15.003 > Starting access point for configuration portal.
23:08:15.005 > SSID: 'espresense-9c4bd5'
23:08:15.008 > [   511][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
23:08:15.016 > [   511][E][WiFiAP.cpp:141] softAP(): enable AP first!
23:08:15.019 > Failed to start access point!
23:08:15.520 > IP: 0.0.0.0
```

I'm a bit new to these ESP32C3 devices so feel free to tell me that this issue is beyond your scope for your project (as it would appear that some portion of memory was not properly set up).  My module has 2MB of flash (rather than 4) and I note that the "esp32-c3-devkitm-1.json" that is referenced from the tasmota platform files notes 4GB.  Would that do it?

The project uploaded to the module just fine - no errors.

Thank you in advance!

DTTerastar, 2022-08-29:
softAP(): enable AP first! is the interesting bit... researching...

DTTerastar, 2022-08-29:
Actually, nm I found a way to autofix this issue.  Let me release the next beta with it and you can test it.

DTTerastar, 2022-08-29:
https://github.com/ESPresense/ESPresense/releases/tag/v3.0.0b11 please test!

shane64, 2022-08-29:
Thank you again.  I've loaded b11 and It appears to produce the same output as follows:

```
15:25:43.379 > ESP-ROM:esp32c3-api1-20210207
15:25:43.389 > Build:Feb  7 2021
15:25:43.389 > rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
15:25:43.394 > SPIWP:0xee
15:25:43.394 > mode:DIO, clock div:1
15:25:43.397 > load:0x3fcd6100,len:0xf4
15:25:43.407 > load:0x403ce000,len:0x8a0
15:25:43.407 > => 0x403ce000: ?? ??:0
15:25:43.407 > load:0x403d0000,len:0x218c
15:25:43.417 > => 0x403d0000: ?? ??:0
15:25:43.417 > SHA-256 comparison failed:
15:25:43.417 > Calculated: 402ae7f8d96ee2a56a5a78dd380569c46de7b91df4dfdb33de30b80f17413178
15:25:43.417 > Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
15:25:43.419 > Attempting to boot anyway...
15:25:43.433 > entry 0x403ce000
15:25:43.433 > => 0x403ce000: ?? ??:0
15:25:43.446 > [    6[    69][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
15:25:43.451 > [    69][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
15:25:43.457 > Setup network
15:25:43.460 > [    81][E][vfs_api.cpp:24] open(): File system is not mounted
15:25:43.465 > [    82][E][vfs_api.cpp:24] open(): File system is not mounted
```
last line repeated many more times
```
15:25:43.760 > [   372][E][vfs_api.cpp:24] open(): File system is not mounted
15:25:43.765 > [   378][E][esp32-hal-i2c.c:87] i2cInit(): i2c_param_config failed
15:25:43.771 > [   384][E][vfs_api.cpp:24] open(): File system is not mounted
```
last line repeated many more times
```
15:25:43.860 > [   473][E][vfs_api.cpp:24] open(): File system is not mounted
15:25:43.865 > [   480][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
15:25:43.871 > [   484][E][vfs_api.cpp:24] open(): File system is not mounted
15:25:43.876 > [   490][E][vfs_api.cpp:24] open(): File system is not mounted
15:25:43.882 > First contact!
15:25:43.885 > 
15:25:43.885 > [   506][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
15:25:43.890 > Starting access point for configuration portal.
15:25:43.893 > SSID: 'espresense-9c4bd5'
15:25:43.896 > [   516][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 4353
15:25:43.904 > [   517][E][WiFiAP.cpp:141] softAP(): enable AP first!
15:25:43.907 > Failed to start access point!
15:25:44.409 > IP: 0.0.0.0
```
Thank you.

DTTerastar, 2022-08-29:
I think you have a 2MB flash instead of the required 4MB flash. :(

DTTerastar, 2022-08-29:
I also saw this on one of my early espc3 and put in this bug/feature req: https://github.com/esphome/esp-web-tools/issues/254

shane64, 2022-08-29:
Yes - as previously mentioned, it's got 2MB of flash.  I had noticed that the "esp32-c3-devkitm-1.json" that is referenced from the tasmota platform files notes 4MB.  Could that be tweaked to 2MB to solve the problem (if that's used to set the formatting parameters) (or does the firmware actually require a 4MB module - which I think you answered in this thread)?

Thank you.

RaoulSargent, 2022-08-30:
**PLEASE PLEASE PLEASE keep a 4MB limit** - or at least a 4MB option without all features.

ESP32 devices are readily available in UK in 4MB versions, but NOT 8MB.

Like Tasmota, perhaps a 'Sensors' version of the firmware, and a 'Standard' version of the software going forward.

DTTerastar, 2022-08-30:
I was mostly joking.  8MB are rare in the US as well :)

PR #489 (2022-06-11): Initial esp32c3 support, also upgrade esp32 framework to 3.5.
Fixes #134
commit d2e4f452 (2022-06-11): Initial esp32c3 support, also upgrade esp32 framework to 3.5. (#489)

* Initial esp32c3 support! (fixes #134)