ESPresense/ESPresense #135: Apple watch was reporting as Close but stopped.

marave, 2021-11-28:
**Describe the bug**
My apple watch was detected and still is but no longer shows as "Close" when held close to the ESP32 device.  My iPhone still works as expected.

**Expected behavior**
When I move my apple watch close to the sensor I expect Close status to be reporting on the serial monitor.

**Logs**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮Connecting to WiFi *******************
Connecting to WiFi *********************
Version:     v2.0.12
IP address:   192.168.0.239
DNS address:  192.168.0.1
Hostname:     espresense-office
Room:         office
Max Distance: 1.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor: 
Connecting to MQTT 192.168.0.130 1883
0 New   | MAC: 63c5c3a0f5d5, ID: apple:1005:9-24                                              2a18b531e3
0 New   | MAC: 7198348eea61, ID: apple:1005:9-24                                              039805ea9a
0 New   | MAC: 78b2e9bfda27, ID: apple:1005:9-12                                              011c1b4fcd
0 New   | MAC: 45b3528ed4b9, ID: apple:1006:10-6                                              371e736b85d6
[32m0 Close | MAC: 45b3528ed4b9, ID: apple:1006:10-6                                              (0.17m) -39dBm[0m
0 New   | MAC: 6b27f86d3973, ID: apple:1005:9                                                 021894d197                                             
[33m0 Left  | MAC: 45b3528ed4b9, ID: apple:1006:10-6                                              (0.19m) -68dBm[0m
[32m0 Close | MAC: 45b3528ed4b9, ID: apple:1006:10-6                                              (0.46m) -33dBm[0m
[33m0 Left  | MAC: 45b3528ed4b9, ID: apple:1006:10-6                                              (0.45m) -61dBm[0m

In the logs the apple:1005:9-24 is my watch and the apple:1006:10-6 is my iPhone.  I swear it was registering the watch as close.

**Hardware Details**
Apple Watch Series 4

**Screenshots**
If applicable, add screenshots to help explain your problem.

marave, 2021-11-30:
Probably expected behavior that I didn't read about in my search but after reloading the ESPresense with an erase and then my watch shows as close again.  My wife's watch has the same ID though so is there anything that can be done about that?

DTTerastar, 2021-12-01:
The above article is basically how ESPresense works.  We look for that nearby info as well.  If you turn on the query option it will also request the model number as well so as long as your wife's watch is a different model it'll be able to tell them apart.

marave, 2021-12-02:
Thanks for the reply.  The answer was right in front of me so sorry to bother you.

closed 2021-12-02 by marave

DTTerastar, 2021-12-02:
No problem.  You can always help by fixing some of the documentation on https://github.com/ESPresense/ESPresense.github.io