ESPresense/ESPresense #139: Support for multi-sensor

circa1741, 2021-12-02:
**Is your feature request related to a problem? Please describe.**
Please see https://www.reddit.com/r/homeassistant/comments/r78ufv/weve_built_multisensor_hardware_specifically_for/. It would be nice for ESPresense to be able to support the different sensors.

**Describe the solution you'd like**
For ESPresense to be able to support the different sensors in the above link.

**Describe alternatives you've considered**
For those of us not into soldering individual components, it would be nice to simply upload the ESPresense in the product above.

DTTerastar, 2021-12-02:
Very cool!  I'd love one of those.  I didn't see any pricing/where to order it.

circa1741, 2021-12-02:
They are still working on the product. However, ESPresense was mentioned in the thread.

corgan2222, 2021-12-04:
#shutupandtakemymoney :)
I would buy a bunch of those too.

closed 2022-04-10 by stale[bot]