ESPresense/ESPresense #140: Restart via mqtt message

bikefright, 2021-12-04:
Firstly, fabulous project!

I'm at the crossroads where I have 4 instances of Andrew Freyer's BT Monitor deployed and in the past 2 days I've deployed 5 x ESP32s with your ESPresense (would have been 6, but one was DOA) with a view to replacing Monitor.  My 3d printer has been busy printing cases for the ESPs 🤓

One feature I like about the other project is the ability to restart an instance by publishing an mqtt message with a specific topic/payload.  I have used this to trigger a restart of my Monitor instances from node-red when various other telemetry indicates there *may* be a problem.  So much more convenient than connecting to each instance individually and issuing the relevant systemctl or reboot commands.

Would you consider similar functionality?

Perhaps a separate topic along the lines of:

espresense/rooms/xxxxx/restart

This would potentially allow you to send a message with a payload of: True/1/Yes to one or all instances (using espresense/rooms/+/restart).

Apologies if this functionality is already available - I think I've checked where I think it would be if it were there already.

Cheers,

Rowland.

corgan2222, 2021-12-04:
I know this problem from other ESP-Bluetooth Projects, like ESP-Keyble.
But I have 5 ESPresensce running, and they never hang or needed to be restarted. Did you ever face this issue?

bikefright, 2021-12-04:
It's early days for me, so please forgive my inexperience with ESPresense - one of the reasons for looking at another solution was a particular key household member's (wife's) iPhone 6s which has proven the most annoying/unreliable device to keep tabs on with Monitor but I know I would have to pry it from her cold dead hands before she'd give it up 😂.

I have a number of automations (heating/lighting/cctv etc.) that rely if not entirely then in part on reliable presence detection, so if I'm working away, the ability to conveniently (and quickly) remotely restart various home automation components (which would potentially now include my ESPresense nodes) is key for me however "blunt" an approach that may seem on the surface.  I can troubleshoot and diagnose when I get back.

I guess being able to restart is (for me) just as potentially important as being able to set the Active Scan, Max Distance and Query config parameters.

Cheers,
Rowland.

closed 2022-04-10 by stale[bot]

halogen28372, 2022-12-05:
I'd like to resurrect this. I have several devices that lose communication with home assistant but can still be reached on the network. Having a way to reboot them automatically would infinitely improve ESPresence for me.

DTTerastar, 2022-12-05:
https://github.com/ESPresense/ESPresense/blob/b9787ba5a05b5964bff0a5e2dab44002ded527e5/src/main.cpp#L307

It does support restart already.

bikefright, 2022-12-05:
As the OP, I can confirm that this feature is now present and I take advantage of it by sending the correct MQTT message to the relevant command topic.

I've attached a couple of screenshots showing the relevant button in HA for one of my espresense devices and another showing the MQTT info - and hence the command topic for the same device.

I had to delete and rediscover my espresense nodes in MQTT not so long ago to realign the discovered topics with reality!  I had an earlier major version and IIRC there were some MQTT topic changes which when espresense updated, HA refused to see.  Simply deleting the devices from within HA, deleting any retained topics using MQTT Explorer (or any other tool to browse/change the MQTT topic tree) and rediscovering resolved this problem for me.

Personally I use nodered for this kind of online/offline automation/checking (I check the status of the device by monitoring the telemetry of all my nodes (the topic to listen on for me is: espresense/rooms/+/telemetry).

I fire these telemetry events through a "timeout" countdown node (one for each espresense node) with a 2 minute countdown.  At 30s remaining, a "marginal" notification is sent. If the 2m countdown expires an "offline" alert is sent and an attempt is made to restart the node using the relevant command topic (for the node below, this is: espresense/rooms/office_atom/restart/set).

Hope this helps 🙂
Rowland.

[image]
[image]