ESPresense/ESPresense #141: Stuck at Scanning for WiFi Networks on webserver

yang3535, 2021-12-05:
**Describe the bug**
I have tried flashing several MH-ET Live Minikit ESP32 boards with v2.0.12.   The first board worked perfectly.  Three others appeared to flash ok (I tried both web based and ESPHome falshers), but when attampting to configure via we browser, all three get stuck at Scanning for WiFi Networks and go no further.

**Expected behavior**
List of scanned WiFi networks for selection

**Logs**
Here is the log from the ESPHome flasher log:
```
Using 'COM10' as serial port.
Showing logs:
[14:12:52]ets Jun  8 2016 00:22:57
[14:12:52]
[14:12:52]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:12:52]configsip: 0, SPIWP:0xee
[14:12:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:12:52]mode:DIO, clock div:2
[14:12:52]load:0x3fff0018,len:4
[14:12:52]load:0x3fff001c,len:1044
[14:12:52]load:0x40078000,len:8896
[14:12:52]load:0x40080400,len:5828
[14:12:52]entry 0x400806ac
[14:12:52]Connecting to WiFi (24:6F:28:13:BF:54)...
[14:12:52]
[14:12:52]Brownout detector was triggered
[14:12:52]
[14:12:52]ets Jun  8 2016 00:22:57
[14:12:52]
[14:12:52]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:12:52]configsip: 0, SPIWP:0xee
[14:12:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:12:52]mode:DIO, clock div:2
[14:12:52]load:0x3fff0018,len:4
[14:12:52]load:0x3fff001c,len:1044
[14:12:52]load:0x40078000,len:8896
[14:12:53]load:0x40080400,len:5828
[14:12:53]entry 0x400806ac
[14:12:53]Connecting to WiFi (24:6F:28:13:BF:54)...
[14:12:53]First contact!
[14:12:53]
[14:12:53]Starting access point for configuration portal.
[14:12:53]SSID: 'espresense-54bf13'
[14:12:54]IP: 192.168.4.1
[14:13:00]ets Jun  8 2016 00:22:57
[14:13:00]
[14:13:00]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:13:00]configsip: 0, SPIWP:0xee
[14:13:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:13:00]mode:DIO, clock div:2
[14:13:00]load:0x3fff0018,len:4
[14:13:00]load:0x3fff001c,len:1044
[14:13:00]load:0x40078000,len:8896
[14:13:00]load:0x40080400,len:5828
[14:13:00]entry 0x400806ac
[14:13:01]Connecting to WiFi (24:6F:28:13:BF:54)...
[14:13:01]
[14:13:01]Brownout detector was triggered
[14:13:01]
[14:13:01]ets Jun  8 2016 00:22:57
[14:13:01]
[14:13:01]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:13:01]configsip: 0, SPIWP:0xee
[14:13:01]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:13:01]mode:DIO, clock div:2
[14:13:01]load:0x3fff0018,len:4
[14:13:01]load:0x3fff001c,len:1044
[14:13:01]load:0x40078000,len:8896
[14:13:01]load:0x40080400,len:5828
[14:13:01]entry 0x400806ac
[14:13:02]Connecting to WiFi (24:6F:28:13:BF:54)...
[14:13:02]First contact!
[14:13:02]
[14:13:02]Starting access point for configuration portal.
[14:13:02]SSID: 'espresense-54bf13'
[14:13:03]IP: 192.168.4.1
[14:13:22][E][WebServer.cpp:633] _handleRequest(): request handler not found
[14:13:23][E][WebServer.cpp:633] _handleRequest(): request handler not found
[14:13:24][E][WebServer.cpp:633] _handleRequest(): request handler not found
[14:13:30]4 WiFi networks found.
```

**Hardware Details**
MH-ET Live Minikit ESP32 board (probably a clone)

**Screenshots**
If applicable, add screenshots to help explain your problem.

ZeddicPrime, 2022-02-16:
This sounds similar to issue #243 I reported a few days ago. I ended up switching to other hardware and so far they seem to be much more stable.

DTTerastar, 2022-02-21:
Can this be closed?

yang3535, 2022-02-22:
Closing this and deferrinto resolution of #243

closed 2022-02-22 by yang3535